z88dk/z88dk #1: Problem with preprocessor of sccz80?

pauloscustodio, 2017-01-15:
When building the package from git with ./build.sh, I get:
```
make -C lib3d
make[1]: Entering directory '/cygdrive/c/Users/paulo/git/z88dk/libsrc/lib3d'
zcc +test -O3 -vn -Wn43  -c *.c
sccz80:"/home/paulo/winhome/git/z88dk/lib/config//../..//include/msx.h" L:341 Warning:#17:Expected ';'
sccz80:"/home/paulo/winhome/git/z88dk/lib/config//../..//include/msx.h" L:342 Error:#45:Missing token, expecting } got \
```

Line 342 of msx.h has ```LINE_T_MEMBERS;```, which is a macro expansion containing multi-lines with backslash-newline. It seems the preprocessor is keeping the backslashes.

suborb, 2017-01-15:
I can't reproduce this one on a Mac so it looks like a windows only problem.

Has msx.h got any funny line endings etc?

pauloscustodio, 2017-01-15:
Yes, it has CR-LF line endings.

The default in git is to commit always with LF endings, but to write the native line ending in the working area (LF on Unix, CR-LF on windows). 

Maybe the C preprocessor is opening the files in binary mode?

closed 2017-01-15 by pauloscustodio

pauloscustodio, 2017-01-15:
Issue solved. 

git uses by default the native line ending in the sandbox. This causes problems with cygwin that declares itself as UNIX, requiring LF-only line endings, but on a Win32 files git saves files with CR-LF.

Solved by declaring to git that all files have LF endings independently of the platform. Microsoft compilers on Win32 handle LF-files well.

commit 6049d6f5 (2017-01-15): Fix #1: use LF-only line endings