z88dk/z88dk #2: Problem building z88 library?

pauloscustodio, 2017-01-15:
When building the package from git (./build.sh), I get:
```
make -C fcntl/z88
...
z80asm -d -I/home/paulo/winhome/git/z88dk/lib/config//../ -DFORz88 -x/cygdrive/c/Users/paulo/git/z88dk/libsrc//z88_clib @./z88.lst
Error: cannot read file 'stdio/obj/fopen_z88'
Error: cannot read file 'stdio/obj/freopen_z88'
2 errors occurred during assembly
make: *** [Makefile:244: z88_clib.lib] Error 1
```

suborb, 2017-01-15:
Thanks Paulo. Those files aren't used anymore, but it looks like I failed to remove them from CVS or the git import skipped that change. We may have a few more of these issues I suspect.

pauloscustodio, 2017-01-15:
I have deleted the lines referring to these files in the several *.lst files. I can commit the change.

suborb, 2017-01-15:
I already did that :) A fresh checkout now builds for me.

pauloscustodio, 2017-01-15:
Works for me too.

closed 2017-01-15 by pauloscustodio