z88dk/z88dk #6: ZSDCC: Bytes encoded as hex chars in character strings cause strings to be improperly generated in memory

aralbrec, 2017-01-16:
Example:

const unsigned char hall_valids[42] = "\x01ABCDEFGHIJKLMNOPQRSTUVWXYZ0123456789.~ {";

zcc +zx -v -a -clib=sdcc_iy test.c

Results in:

_hall_valids:
    .db 0xef
    .ascii "GHIJKLMNOPQRSTUVWXYZ0123456789.~ {"
    .db 0x00
    .db 0x00
    .db 0x00
    .db 0x00
    .db 0x00
    .db 0x00
    .db 0x00

The 0x01 byte that should be there is replaced by 0xef and the following A-F chars are not present.

If the byte 0x01 is instead encoded in octal, the string is fine.

This is a problem within sdcc that I've tried to push on them :p
https://sourceforge.net/p/sdcc/bugs/2577/

In my imagination this shouldn't be hard to fix so if the sdcc team don't solve it soon maybe we can have a look.

pauloscustodio, 2017-01-16:
The implementation is correct as far as the C-standard is concerned, see
"since a hexadecimal escape sequence is terminated only by a non-hexadecimal character" (http://port70.net/~nsz/c/c89/c89-draft.txt?bcsi/ac-c2f74769f0d2d0ad=27736B8600000603IRdhDdlO779YawGmgsVlIVnIx15HAgAAAwYAAO0lIgAIBwAAAwAAAH//BQA=)

The actual value is implementation dependent, in this case a modulo-256 of the hex number 01ABCDEF is taken, resulting in EF hex.

closed 2017-01-16 by pauloscustodio

suborb, 2017-01-16:
For reference, sccz80 has different behaviour and truncates to a byte (which I think is what Alvin expected).

Clang reports an error (hex escape sequence out of range) and gcc 4.1.2 reports the same as a warning.

So although sdcc is technically correct it's not being particularly helpful.

aralbrec, 2017-01-16:
Yes following Paulo's response I also looked it up in a more recent C11 standard document and sure enough the hex or octal constant is taken to be the longest string of valid hex or octal digits following the initial \x or \0.  I had always been under the impression that exactly two hex digits are taken and exactly three octal digits are taken because that's what's required to represent an 8-bit char.

But we have to keep in mind there is unicode now (obviously I don't keep that in mind) so now I'm wondering if this is something that changed in the past or if I just never had it right.

So clang and gcc are correct -- I'm trying to encode a hex character code > 255 into a string containing 8-bit chars.

I suppose the correct thing to do is:

const unsigned char hall_valids[42] = "\x01" "ABCDEFGHIJKLMNOPQRSTUVWXYZ0123456789.~ {";

I've got code out there written in a professional capacity that will be wrong too, although I think it's accidentally correct as the hex encodings tend to be \xaa\xaa\xaa... with the backslash saving the day.

aralbrec, 2017-01-16:
Maybe this is actually a bug report on sccz80 then :D

suborb, 2017-01-16:
Well, the bug report would be undefined I guess. I've added a warning in now. The behaviour hasn't changed, it still takes two hex characters and forms a byte.

commit e0d06d6d (2017-01-16): Fix #6: Warn when an overlong hex constant is used.

Our implementation of undefined is to to take two bytes of it.
commit 8b6a3147 (2017-01-16): Fix #6: Warn when an overlong hex constant is used.

Our implementation of undefined is to to take two bytes of it.