z88dk/z88dk #7: Clone doesn't pull request well.

feilipu, 2017-01-17:
`git clone https://github.com/RC2014Z80/z88dk.git`

change 3 files...

`git add .`
`git commit -m "rc2014 acia - optimisedtx"`

[Showing  217 changed files  with 22,755 additions and 22,745 deletions](https://github.com/RC2014Z80/z88dk/commit/1624e4582c18149cf09d1316f9948a56cda3dff5).

WTF!!! Where did all this garbage come from?
And, how can I make a clean pull request for you?

Phillip

suborb, 2017-01-17:
I think this is a line ending issue. @pauloscustodio changed the default line ending which has caused a bit of chaos due to the mixed endings before. All of the irrelevant changes are whitespace. To avoid your pull request having all the change, just commit the files that you changed.

We need to sort this out and as we do the size of your PR will decrease. Otherwise just add the files you explicitly changed to the commit.

feilipu, 2017-01-17:
[Done.](https://github.com/z88dk/z88dk/pull/8)

closed 2017-01-17 by feilipu