z88dk/z88dk #9: sccz80: Support __z88dk_fastcall and __z88dk_callee

suborb, 2017-01-17:
In sdcc these are defined after the function prototype, in sccz80 they are defined between type and function name.

In the classic library, there's some macro magic in sys/compiler.h to get this to work if the function is annotated with both versions.

However, it's a bit messy, so allow sccz80 to accept the modifiers after the prototype.

Additionally, swallow __smallc which is the sdcc indicator that the calling convention is l->r

closed 2017-01-17 by suborb

aralbrec, 2017-01-18:
Can you get sccz80 to ignore __preserves_regs(...) ?  That would complete compatibility with sdcc.
extern void srand_fastcall(uint16_t seed) __preserves_regs(b,c,d,e,h,l) __z88dk_fastcall;

suborb, 2017-01-18:
I've added that now.

As a thought for the new library it should be possible to drop __LIB __ as well - all that does these days for the new library is to remove the _ prefix.

For classic we need __LIB__ for varargs functions given the calling conventions are different - in classic the same library is used for both sdcc and sccz80 so it needs to have different names. However I may start removing it - the main reason for the difference was to work around z80asm and XLIB/XREF/XDEF differences.

Whilst doing this I found an obscure bit of sccz80 syntax that I obviously needed at one point (but forgot about):

int function(int arg1, char *arg2) @ 40000;

Which prototypes a function to be at that address.

aralbrec, 2017-01-18:
That should be "__preserves_regs(...)" with an "s" on "preserves".

> Whilst doing this I found an obscure bit of sccz80 syntax that I obviously needed at one point (but forgot about):
> int function(int arg1, char *arg2) @ 40000;

Yes I even wrote a little about it:
https://www.z88dk.org/wiki/doku.php?id=libnew:target_embedded#absolute_placement_of_code_or_data_in_memory

but I had problems in actually using it since sccz80 ignores the "static" keyword when associated with it.  I'd wanted to put:

`static unsigned char NIRVANAP_isr[3] @ (56698+328*NIRVANAP_TOTAL_ROWS);`

into a header file but because "static" is ignored, this generates "public" definitions of this constant in different source files resulting multiple instances of "_NIRVANAP_isr" at link time.

Instead I prefer the c/asm combination mentioned in the above link.  The prototype is done in C as extern and then the asm just defines the label to be the constant address.

suborb, 2017-01-18:
Ouch, sorry about that I'll fix that one tonight.

The function variant is kept local to the file so it ignores the scope in a different way.  The variable variant does make things extern. Whilst I'm down in the weeds I can make the variable version also keep them file local for consistency.

aralbrec, 2017-01-18:
> As a thought for the new library it should be possible to drop __LIB __ as well - all that does these days for the new library is to remove the _ prefix.

I suppose as you said it's useful now to distinguish entry points for sccz80 and sdcc if object files generated by the two compilers can be mixed.  This would require some amount of editing in the new c lib so I'll make a note of it for the source code reorganization.

The __z88dk_fastcall / __z88dk_callee accommodation in sccz80 is less problematic in the new c lib since it generates separate header files for each compiler from a common prototype.  But I think users especially will appreciate less typing.

suborb, 2017-01-18:
What are the reasons for generating a separate headers? I'm guessing it was the annotations and function pointers?

Less typing is definitely welcome, I wish I'd thought of it when I was making the _classic_ library work with classic.

Mixing object files is known to work with _classic_ (given that part of the C library is compiled C and you can compile user code with sdcc), but I'm not sure how it would work with _new_ given the ix/iy variants.

aralbrec, 2017-01-18:
> What are the reasons for generating a separate headers? I'm guessing it was the annotations and function pointers?

Yes it's down to some differences, the biggest being the attributes.  This will solve the sccz80/sdcc differences in attributes but there is also a set of clang headers in there now**.  Clang needs completely plain headers with all attributes removed.  If some other compiler comes along, it may also be best accommodated by being able to generate headers with its own set of attributes for it.

The other differences:

* How function calls through pointers to fastcall functions are handled.  In sccz80 we made the decision a long time ago that generated calls through function pointers would also work for fastcall functions by agreeing to put the last parameter pushed in DEHL (does it work for > 16 bit? I'm not sure).  So there is no distinction in type between a pointer to a function and a pointer to a fastcall function and in sccz80's headers we don't do the callee trick for fastcall functions.  sdcc, OTH, makes no such accommodation so either all function pointer calls use standard linkage or the function pointer is typed based on fastcall/callee linkage.  Right now sdcc has fastcall included in a function pointer's type (and callee is unknown to the compiler).  So you will see in the new c lib, the callee trick is also applied to fastcall for sdcc so that function pointers always use standard linkage and the user does not have to worry about typing for pointers to library functions.  For mixing object files, we may have to homogenize this difference too.

* sdcc has a 64-bit integer type so some headers add 64-bit related functions and some define different integer-limit related things for sdcc only.  I think I did a lot of these as "#ifdef __SDCC" rather than only having it appear in sdcc-specific headers.

* sdcc aliases float as double with warnings and sccz80 aliases double as float.  To get rid of warnings, the standard actually introduced new types "float_t" and "double_t" so these are defined differently for sdcc and sccz80 in headers, again I think this was done with "#ifdef __SDCC" and "#ifdef __SCCZ80".  If those types are used for floats then source code is compatible and no warnings will occur when using float_t or double_t.   Philip at sdcc has mused about supporting different float types in sdcc so maybe new differences will come about in the future and I'm not sure if that will require different attributes or headers.  In terms of object file compatibility there is a big problem since sdcc floats are 32-bit and sccz80's are 48-bit.  The math48 library solves the difference by inserting conversion code between sdcc and the math lib.  In C source I think the only way may be to have the user be aware when floats are passed between functions compiled by different compilers and have him insert calls that do the conversion explicitly as floats cross compiler boundaries.

** I am successfully compiling with clang->sdcc with a little manual intervention; the code is not as good as regular sdcc but I see this as a possible route to compiling other languages.

aralbrec, 2017-01-18:
> Mixing object files is known to work with classic (given that part of the C library is compiled C and you can compile user code with sdcc), but I'm not sure how it would work with new given the ix/iy variants.

I'm not sure yet either.

aralbrec, 2017-02-17:
There is one more sdcc attribute that could be swallowed by sccz80:

```
void myfunc(int a) __naked
{
__asm
...
__endasm;
}
```

The `__naked` attribute indicates that the compiler should not add prologue or epilogue code to the function.  It's widely used and intended for functions written entirely with inlined asm.

suborb, 2017-02-17:
I'm not sure that supporting __naked is sensible given the parameter pushing differences - I'd much prefer an error to be reported rather than incorrect code be generated.

aralbrec, 2017-03-31:
I'm running into this again.

__naked is frequently used in sdcc source where assembly is wrapped inside a c function.  It's preventing sccz80 from compiling the same source without adding "-D__naked=" to the compile line.  If it's swallowed (and ignored) by sccz80, the source code is compatible.

suborb, 2017-09-03:
Cleaning up time.

__naked was actually added in 64d8d6918a52f2a1289696ee1fd26ea9ef3da4d0 Issue #139

commit 67327ca7 (2017-01-17): Fix #9: Make sccz80 accept the sdcc __smallc, __z88dk_fastcall,
__z88dk_callee modifiers.

This should simplify header files a little.
commit 2ceb9b62 (2017-01-18): Fix #9: Make sccz80 swallow __preserve_regs() as well to get better
sdcc compatibility.
commit a37dea23 (2017-01-18): Fix #9: Typo on preserves_regs