z88dk/z88dk #15: z88dk: Allow object files generated by zsdcc and sccz80 to be mixed in the same binary

aralbrec, 2017-01-18:
The user may want to choose which compiler is used to compile .c source on a per-file basis. 

Some reasons:

* One compiler may do a better job in terms of execution speed.
* One compiler may do a better job in terms of code size.
* One compiler has more bits in its float type.
* One compiler may be more desirable to use during development due to compile speed.

In the documentation it's been mentioned that due to various differences, a project must be completely compiled by one compiler or the other, ie due to these differences the object files are not compatible.

This is not quite correct.  The classic c library can mix zsdcc and sccz80 object files in the majority of circumstances whereas the new c library cannot mix them in most circumstances.  Also, the c libraries themselves must be aware of the mixing so that library code is shared between the compilers rather than duplicated in output binaries in order for this to be tenable.

This topic is here to act as reminder to the devs and collect relevant information.

Some discussion on this has already occurred in #9.

aralbrec, 2017-01-18:
Another point of difference is in how char parameters are push onto the stack.  sdcc will push one byte for each char whereas sccz80 will push a word.

In a recent project https://bitbucket.org/CmGonzalez/pietro_bros/ changing the interface to the nirvana+ library to use chars in zsdcc builds saved 400 bytes.

pauloscustodio, 2017-01-18:
Just thinking out loud: could we add attributes to function entry points and change the linker to search a matching function name and function attributes? This would require changes in the compilers and assembler, to pass the attributes to the object file, and in the linker to search for a matching function.

aralbrec, 2017-01-20:
A simpler variation on that is to get the preprocessor to substitute different names for function entry points depending on which compiler is running:

#ifdef __SDCC
#define strcpy sdcc_strcpy
#end if

#ifdef __SCCZ80
#define strcpy sccz80_strcpy
#end if

Then have C entry points for each that collect parameters and then jump into the common asm implementation.

So the compiler and attributes get encoded in the name rather than having to modify any of the toolchain.

aralbrec, 2017-01-22:
A note regarding function renaming for zsdcc.  sdcc will inline a handful of string functions:

>    #ifndef __SDCC_DISABLE_BUILTIN
> 
>    #undef memcpy
>    #undef strcpy
>    #undef strncpy
>    #undef strchr
>    #undef memset
> 
>    #define memcpy(dst, src, n)          __builtin_memcpy(dst, src, n)
>    #define strcpy(dst, src)             __builtin_strcpy(dst, src)
>    #define strncpy(dst, src, n)         __builtin_strncpy(dst, src, n)
>    #define strchr(s, c)                 __builtin_strchr(s, c)
>    #define memset(dst, c, n)            __builtin_memset(dst, c, n)
> 
>    #endif

This replaces the macro magic we use for fastcall/callee linkage in favour of selecting sdcc's built-ins for these functions.  In practice, sdcc's inlining of these functions are a big win in terms of speed.

Sometimes sdcc is unable to inline these functions and instead the compiler will insert a call to the c library's implementation.  However, this insertion happens after the preprocessor has already run so we do not get callee/fastcall linkage selected as we would want.

Instead, this case is caught by [copt](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_opt.2) in post-processing along with the other calls that sdcc makes to its support functions.

If we adopt naming as suggested in the last post, the case of built-ins must be kept in mind.