z88dk/z88dk #16: z80asm: cannot specify name of generated object file

pauloscustodio, 2017-01-18:
from:	alvin (alvin_albrecht@hotmail.com) <lists@suborbital.org.uk> via lists.sourceforge.net 
to:	z88dk-developers@lists.sourceforge.net
date:	Sun, Oct 23, 2016 at 9:03 AM

When assembling to a single object file there is no way to control the output filename.

z80asm thing.asm
will always generate thing.o

I'd like to be able to do this:

z80asm thing.asm -o c:\temp\..\zcc0000.o

Line 39 of options_def.h hints that this was considered but never implemented:
OPT_VAR( char *,    consol_obj_file, NULL)      /* set by -o and no -b */

Because z80asm will search for includes from the .asm file's location, zcc needs to assemble .asm source files in their original source directories and we don't want to pollute the source directories with temporary files.

pauloscustodio, 2017-01-18:
from:	Paulo Custodio <pauloscustodio@gmail.com>
to:	"alvin (alvin_albrecht@hotmail.com)" <lists@suborbital.org.uk>
cc:	z88dk dev <z88dk-developers@lists.sourceforge.net>
date:	Sun, Oct 23, 2016 at 4:51 PM

>>Line 39 of options_def.h hints that this was considered but never implemented:
>>OPT_VAR( char *,    consol_obj_file, NULL)    /* set by -o and no -b */
>The line you refer was recently introduced to support the consolidated object file; the -o option for naming the binary is used to name the consolidated object, which is produced at the link phase by reading all the object files from the input assembly files.
>As z80asm does process a list of asm files in  one go, e.g. to link a binary or build a library, the object file has to be determined from the source name.
Another possibility is to distinguish between two cases:  if there is only one source file use the output filename.  If there are many source files, ignore the output filename.  Except for the consolidated object, libraries and binaries which are intended to combine many files into one.

>The problem you mention: could it be solved by passing an extra -I option to z80asm with the original source directory of the source?
For now, zcc is copying .asm files to the temp dir before assembling so the .o is created in the temp dir.  The include path will not resolve properly if the .asm file was not originally sitting in the cwd but nobody's complained about this in the last decade at least :P

But yes that can be done.  There is a second issue with this where we've already created the include path string that is properly sequenced in the order the user wants.  So adding a new include path at the point of assembling won't necessarily honour that sequence without some other changes.

pauloscustodio, 2017-01-18:
from:	alvin (alvin_albrecht@hotmail.com) <lists@suborbital.org.uk> via lists.sourceforge.net 
to:	z88dk-developers@lists.sourceforge.net
date:	Sun, Oct 23, 2016 at 11:21 PM

> The line you refer was recently introduced to support the consolidated object file; the -o option for naming the binary is used to name the consolidated object, which is produced at the link phase [...]
Another possibility is to distinguish between two cases:  if there is only one source file use the output filename.  If there are many source files, ignore the output filename.  Except for the consolidated object, libraries and binaries which are intended to combine many files into one.

>The problem you mention: could it be solved by passing an extra -I option to z80asm with the original source directory of the source?
For now, zcc is copying .asm files to the temp dir before assembling so the .o is created in the temp dir.  The include path will not resolve properly if the .asm file was not originally sitting in the cwd but nobody's complained about this in the last decade at least :P

But yes that can be done.  There is a second issue with this where we've already created the include path string that is properly sequenced in the order the user wants.  So adding a new include path at the point of assembling won't necessarily honour that sequence without some other changes.

aralbrec, 2017-01-20:
The suggested solution was applied in zcc, that is the user's asm file is copied to the temp directory and then is assembled to object file with the path to the original location added at the end of z80asm's include path list.

But as you say, this may mess with the user's intended path search order.

pauloscustodio, 2017-01-20:
The current z80asm also has already this solution in place: the directory of the main assembly file, i.e. the one mentioned in the command line, is appended to the end of include path before parsing the file and removed afterwards. This allows relative include paths out-of-the-box.

pauloscustodio, 2022-01-08:
Issue solved with the introduction of consolidated object files:
```
z88dk-z80asm -Otest.dir -ozcc0000.o test.asm
```
-Otest.dir creates all object files in that directory.
-ozcc0000.o creates a consolidated object file containing the test.o object also in the test.dir directory.

closed 2022-01-08 by pauloscustodio