z88dk/z88dk #20: z80asm: DEFL - Future incompatible change in z80asm

pauloscustodio, 2017-01-18:
from:	Paulo Custodio <pauloscustodio@gmail.com>
to:	"z88dk-developers@lists.sourceforge.net" <z88dk-developers@lists.sourceforge.net>
date:	Tue, Jan 27, 2015 at 10:09 PM

I intend to implement macros in the assembler, to allow, for example, a linked list to be created at assembly time. In Z80-assembly syntax of other assemblers:

```
link    DEFL 0                 ; define link as a variable, initialize to zero

; macro to create objects
object MACRO #arg1,#arg2
          DEFW  link             ; point to previous object
link     DEFL   ASMPC-2    ; change link to point to this object
          DEFB  #arg1,#arg2  ; define the object data
          ENDM

          object 1,2               ; call macro to create object 1
          object 3,4               ; and object 2

head: DEFW link               ; create pointer to head of list
```

Apart from the macro syntax, I need to have DEFL to define a symbol that changes its value during assembly ("link" in the example above). DEFL and EQU are the usual keywords for constant and variable symbols in other assemblers.

Today DEFL is used to define a double-word (4 bytes) in just a couple of places in the library.

PROPOSAL:
- Create a new DEFDW to be used to define double-word values
- Replace the few DEFL to DEFDW in the library
- Create DEFL with the semantic shown above.

Any comments?

pauloscustodio, 2017-01-18:
from:	Stefano Bodrato (stefano_bodrato@hotmail.com) <lists@suborbital.org.uk> via lists.sourceforge.net 
to:	z88dk-developers@lists.sourceforge.net
date:	Wed, Jan 28, 2015 at 6:41 PM

>Any comments?

I think it makes sense.

pauloscustodio, 2017-01-18:
from:	alvin (alvin_albrecht@hotmail.com) <lists@suborbital.org.uk> via lists.sourceforge.net 
to:	z88dk-developers@lists.sourceforge.net
date:	Wed, Jan 28, 2015 at 9:04 PM

>Today DEFL is used to define a double-word (4 bytes) in just a couple of
>places in the library.
>
>PROPOSAL:
>- Create a new DEFDW to be used to define double-word values
>- Replace the few DEFL to DEFDW in the library
>- Create DEFL with the semantic shown above.

I think that's ok too.

Maybe you'd want to keep the syntax consistent with defc?

defl link = ASMPC - 2

I'm also wondering if there is an advantage to not naming the parameters and instead referring to them by position.  You might be able to shift them, process them recursively, pass them to other macros as done in shell scripts?  Or both could be done with position reference always valid.

MACRO object

   defw link
   link defl ASMPC - 2

   defb $#0, $#1

ENDM

In that specific example:

>; macro to create objects
>object MACRO #arg1,#arg2
>DEFW  link             ; point to previous object
>link     DEFL   ASMPC-2    ; change link to point to this object
>DEFB  #arg1,#arg2  ; define the object data
>ENDM

ASMPC is an assemble time constant so it's an offset from the current section (I was bitten by that earlier as you know :) ) and this won't work unless the current section is ORGed at 0.

It will work if you can generate labels by concatenation:

prev_0:  defw 0

link defl 0

; macro to create objects
object MACRO #arg1,#arg2

prev_##(link+1):  DEFW  prev_##link
DEFL  link = link + 1 ; change link to point to this object
DEFB  #arg1,#arg2  ; define the object data

ENDM

head: defw prev_##link

Definitely not as clean (I would probably pass the base label name as a parameter) but probably unavoidable too.

feilipu, 2019-05-21:
Found DEFQ to work. Although that is not referenced here.

pauloscustodio, 2019-05-22:
Correct. DEFQ replaced DEFL.
The documentation is way behind.

closed 2021-05-01 by suborb