z88dk/z88dk #30: github: create a place to download windows and osx binaries from?

aralbrec, 2017-01-20:
For people who are using github rather than downloading the nightly packages should we be offering a place to download the windows and osx binaries from?

suborb, 2017-01-20:
I'd prefer not to have any auto-commits going on, especially not on binary files.

Perhaps the easiest is to clone and also download the nightly. The binaries can then be extracted from the nightly and put into place?

pauloscustodio, 2017-01-20:
I would prefer to not have any binaries in the z88dk git repository. We can keep the nightly builds as now, and just add a URL to the README at the top of the repository.

I also do not like to have the z80 libraries committed to git - they can be generated from source and appear to git status always as modified, cluttering the development environment. Can we store a them together with the binaries and add a URL to the README also?

aralbrec, 2017-01-22:
Ok let's remove that stuff but we'll have to add a simple write-up in the readme at the same time.

aralbrec, 2017-01-23:
Have a look at the new readme.  Once some brief instructions for a github install are added I think we can get rid of all z80 binaries in the repository.

suborb, 2017-01-23:
The readme looks good, lets kill the libraries.

closed 2017-01-24 by aralbrec

aralbrec, 2017-01-24:
It looks like that might wipe out your local copies of the libs.  You may have to re-acquire them as detailed on front page.

commit 0a65bd8a (2017-01-24): Fix #30 Remove z80 libraries from tracking