z88dk/z88dk #31: github: create a place to hold technical docs?

aralbrec, 2017-01-20:
Recently there was a merge request to remove pdfs for the z80 and mc6850 from the rc2014 target within the new c lib because they were commonly available and occupied a lot of space.  There are more pdfs in the repository having to do with the C standard.  

I'm wondering if we should have a separate repository to collect such things?  They are relevant to drivers and whatnot that are being written as a part of z88dk.

feilipu, 2017-01-20:
It would be good to centralise all documents of a general nature into a separate repository.

Specific z88dk documents (architecture, design, implementation, etc.) should remain within a docs directory within the main repository, for easy reference. 

I'd class C specification docs as general and move them out, for example.

aralbrec, 2017-01-20:
Yes I think I would agree with that and I think having a repository for relevant pdfs would be helpful.  Maybe something appropriately named on https://github.com/z88dk ?

pauloscustodio, 2017-01-20:
I agree.

Name proposal: docs?

aralbrec, 2017-01-22:
techdocs?
docs might mislead people into thinking it's documentation about z88dk itself.

feilipu, 2017-01-22:
techdocs  +1

pauloscustodio, 2017-01-22:
Closed. Created a techdocs repository and moved the Z80 CPU User Manual from the z80asm repository to the new techdocs.

closed 2017-01-22 by pauloscustodio