z88dk/z88dk #34: z88dk: zsdcc and zsdcpp do not run on windows xp?

aralbrec, 2017-01-20:
I have a report that win xp claims that the zsdcc and zsdcpp binaries are not 32-bit.

However, if I run gnu file on them I get this:
zsdcc.exe; PE32 executable for MS Windows (console) Intel 80386 32-bit

I'll doublecheck that the VS2015 settings used to compile these don't have some os version after WinXP ticked someplace.

suborb, 2017-01-21:
Possibly missing the vs2015 redistributable libraries?

aralbrec, 2017-01-22:
Targetting XP with VS tools is becoming a problem as Microsoft no longer supports XP.  I did find some checkboxes in VS2015 that are supposed to allow generation of binaries for XP and I've rebuilt zsdcc & zsdcpp to try it out:

https://drive.google.com/file/d/0B6XhJJ33xpOWUFZKQkhKVlFIUDA/view?usp=sharing

So far I've heard back that the binaries now run under XP but they don't work and stop with errors.  I'm still waiting to hear exactly what errors those are.  It's a bit hard to debug this way so I'm trying to find an XP machine someplace to try myself; unfortunately I don't have easy access to one anymore.

I may have to go back to VS2010 and see if I can still build sdcc with that and target XP with that.  He has no problems running the other z88dk binaries which you are generating via cross-compile.

aralbrec, 2017-01-25:
I've managed to source two xp machines, neither of which work.  Does anyone have easy access to an xp machine anymore?

stefandrissen, 2017-01-27:
I have an XP machine, but I have no quick compile set.

zsdcc.exe shows all the options
zsdcpp.exe does nothing, if I ctrl+c it it then displays:

    # 1 "<stdin>"
    ^ C

whereas if I do this on my Windows 10 machine it displays:

    # 1 "<stdin>"
    # 1 "<builtin>"^C

This may be a simple difference in ctrl+c handling, let me know if you have any files I should throw at them.

aralbrec, 2017-01-27:
> This may be a simple difference in ctrl+c handling, let me know if you have any files I should throw at them.

Thanks for helping out @stefandrissen .

I've updated the same zip to include a pre-digested [eliza.c](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/eliza.c) after all the includes have been processed.

Let's see if we can translate that to assembler with zsdcc:

`zsdcc -mz80 -S --reserve-regs-iy eliza.c`

The output should be "eliza.asm".  This is a raw asz80-syntax file without any post-processing by z88dk (obviously :) ).

If that worked, out of curiosity maybe try with optimization turned up:

`zsdcc -mz80 -S --reserve-regs-iy eliza.c --max-allocs-per-node200000`

Mainly I'm wondering about compile time on an xp-era machine.  My windows 8.1 laptop takes something less than 2 minutes for this.

Let's see if zsdcpp can process a file:

`zsdcpp eliza.c`

This should simply copy eliza.c to the output since the preprocessing has already been done.

If all that works, I think we have working binaries for xp.  The last step is to see if it all comes together with a [z88dk install](https://github.com/z88dk/z88dk) and test compile.

```
cd z88dk/libsrc/_DEVELOPMENT/EXAMPLES
zcc +zx -v -SO3 -startup=8 -clib=sdcc_iy --max-allocs-per-node200000 eliza.c -o eliza -create-app > out.txt
```

You can eliminate the max-allocs-per-node to speed up the compile.

I added "-v" so we can capture the steps taken by zcc in the text file "out.txt" and see where things go wrong if something goes wrong.

stefandrissen, 2017-01-27:
@aralbrec zsdcc is failing directly and not picking up the input file:

    D:\z88dk>zsdcc -mz80 -S --reserve-regs-iy eliza.c
    zsdcpp.exe: fatal error: eliza.c: Invalid argument
    compilation terminated.
    at 1: warning 190: ISO C forbids an empty source file

Note that this is a completely empty pc z88dk-wise - so no environment variables are set - nor is the rest of z88dk installed - so I only have the two executables from your zip.

stefandrissen, 2017-01-27:
Ok, extracted nightly windows build to c:\z88dk - verified not a win32 app - put modified .exes over bin, different result::

    C:\z88dk\bin>zsdcc -mz80 -S --reserve-regs-iy eliza.c
    zsdcpp.exe: error: C:/z88dk/bin/../include: Invalid argument
    zsdcpp.exe: fatal error: eliza.c: Invalid argument
    compilation terminated.
    at 1: warning 190: ISO C forbids an empty source file

Once again - still no environment variables set.

aralbrec, 2017-01-27:
I was thinking zsdcc could be run in isolation from zsdcpp but those errors are about zsdcpp so it seems not.  It might need another switch maybe c1mode or something but let's forget about that since you have a z88dk package downloaded now.

Start another command shell (I'll put z88dk/bin at the front of PATH in case there is anything else with the same binary names installed):

```
set ZCCCFG=c:\z88dk\lib\config
set PATH=c:\z88dk\bin;%PATH%
cd c:\z88dk\libsrc\_DEVELOPMENT\EXAMPLES
zcc +zx -v -SO3 -startup=8 -clib=sdcc_iy --max-allocs-per-node200000 eliza.c -o eliza -create-app > out.txt
```
and let's see if this gets anywhere.  If it does maybe check that m4 also works on xp: `m4 --help`.  We don't compile this one - it's from a gnu windows binary distribution.

Again thanks for trying this.  It's looking like neither of these binaries can parse the command line properly.  I see a few questionable things in the source (re: xp) having to do with signal registration and use of dup2 but I do know that sdcc's own binary builds work on xp so any difficulty must be in the compiler we're using (vs2015).

stefandrissen, 2017-01-27:
C:\z88dk\libsrc\_DEVELOPMENT\EXAMPLES>zcc +zx -v -SO3 -startup=8 -clib=sdcc_iy --max-allocs-per-node200000 eliza.c -o eliza -create-app > out.txt
    zsdcpp: error: .: Invalid argument
    zsdcpp: error: c:/z88dk/lib/config/../..//include/_DEVELOPMENT/sdcc: Invalid argument
    zsdcpp: fatal error: eliza.c: Invalid argument
    compilation terminated.

And the contents of out.txt:
    type "c:\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\zx\zx_crt.asm" >> "d:\temp\<user>\zccB7303.asm"
    
    PROCESSING eliza.c
    zsdcpp -D__Z88DK -DZ80 -DSPECTRUM -D__SPECTRUM__ -D__SPECTRUM -D__SDCC -D__SDCC_IY  -DZ88DK_USES_SDCC=1 -I"." -I"c:\z88dk\lib\config\..\..\\include\_DEVELOPMENT
\sdcc"   "eliza.c" "d:\temp\<user>\zccB7302.i2"

aralbrec, 2017-01-27:
Yes the -I... seems to be fubared.

Let's try one more time, download the zip again - I put a new version of zsdcpp in there.  See if we can get past that at least.

stefandrissen, 2017-01-27:
Got it - is it correct that it is now 157k (was: 441k)?
zsdcc.exe has now been burning through one core for 5 minutes - how long is this supposed to take?

aralbrec, 2017-01-27:
Yes it's compiled with mingw instead of vs2015 and now it's a lot smaller :)

zsdcc can be slow at high optimization.  On my modern machine between 1 and 2 minutes for that program.  If it's taking too long try reducing the optimization by lowering the --max-allocs-per-node number or eliminate that option altogether (default is 3000, best code needs at least 200000).

stefandrissen, 2017-01-27:
Aha. 

Just executed the compile with the nightly on my normal pc - 30 seconds.
WinXP compile has now consumed 15 minutes of CPU time and counting... :-)

aralbrec, 2017-01-27:
>Aha. Doesn't using mingw defeat the purpose of trying to get vs2015 to compile binary that will work on XP? Or is this just to check that zsdcc.exe will work, delaying zsdcpp for later?

Yeah just to see if zsdcc is working.  I don't want to mess too much with sdcc's codebase just to accommodate VS2015.  Maybe we can say vista and later for VS2015 and supply binaries compiled with mingw that will work for xp and later.

> Just executed the compile with the nightly on my normal pc - 30 seconds.
> WinXP compile has now consumed 15 minutes of CPU time and counting... :-)

A few of the examples take a long time at high optimization, up to 15 minutes on my laptop.  If that compile is actually compiling and not spinning, doesn't bode well lol

stefandrissen, 2017-01-27:
OK, after roughly 30 minutes (!!!) it's completed and the .tap file can be loaded into fuse... :-)

stefandrissen, 2017-01-27:
[image]

aralbrec, 2017-01-27:
ROFL, thank you @stefandrissen .  So it's just zsdcpp that is the problem.

I'll have to take a look at the compiler settings to see if zsdcc can be sped up at all.  That's an achingly long time for an 8-bit compiler.  If we can't speed it up maybe we can try mingw builds or ask @suborb to attempt to cross-compile zsdcc with the rest of the win32 binaries and see if they are faster.

stefandrissen, 2017-01-28:
Honestly - I wouldn't bother with XP support. It's nearly three years past it's end of support and I felt very adventurous hooking this machine up the internet...

aralbrec, 2017-01-28:
It is old but I think it still sees use in eastern europe, middle east, africa, ...   This particular report came from eastern europe.

sdcc claims its binaries run on Windows 95 and later.  They compile with mingw and honestly a console application should be able to run in any Windows 95+ environment.

aralbrec, 2017-01-28:
> BTW not sure if it's remote desktop + fuse, but the keyboard response is very knackered.

The key response should be good.  It is not tied to the 50Hz interrupt so will feel different from normal zx input but also should be more responsive.

Maybe check against your Win10 machine?

closed 2017-01-28 by aralbrec

stefandrissen, 2017-01-28:
I think the XP machine was still recovering from overheating after the record breaking compile :-) Via remote desktop it is now workable (not good, but not as wiped out as it was last night). On the physical keyboard of the XP machine it is now acceptable. I have to admit not having attempted typing full sentences into a Speccy for a long long time. :-)

aralbrec, 2017-01-30:
@suborb 

Can we update the windows executables to these:
https://drive.google.com/file/d/0B6XhJJ33xpOWUFZKQkhKVlFIUDA/view?usp=sharing

These will be XP and higher compatible.  XP compiles using zsdcc are super slow but there are people still using XP and willing to put up with the speed.  Apparently the issue is that Windows memory management (malloc/free) was improved on Vista and later and that's why there is a step change in performance.

A note:  VS2015 cannot compile a working zsdcpp for XP.

suborb, 2017-01-30:
Tonight's build will have those binaries.