z88dk/z88dk #40: zcc: When linking multiple files zcc_proj.lst is left behind

suborb, 2017-01-23:
The file is in the source code directory and seems to be leftover from something?

When linking z80asm actually uses a different file that was written at the same time.

aralbrec, 2017-01-24:
zcc_proj.lst isn't used by zcc as you found -- it's information only.  If you're making a big project all on one line or making a library it can show you what files went into building it.

eg, if you go to z88dk/libsrc/_DEVELOPMENT and build one of the libraries with zcc:

`zcc +zx -vn --lstcwd -clib=sdcc_iy -Ca-D__SCCZ80 -x -o test @target/embedded/library/embedded_sccz80.lst
`
The zcc_proj.lst at the end will tell you the full path of every file put into the library.

If no one finds this useful we can get rid of it.  Maybe we could qualify its creation by number of source files so it's only there when likely needed?

btw, with zcc being able to build libraries out of pretty much anything, it could be used to process .lst files containing c source as well as asm mixed together.   Right now the classic lib is built in two steps, compiling the c source files into object separately from generating the library from the object and asm files.  In fact the entire makefile could be replaced by zcc compiling .lst files for each target.   I don't know if you'd want to do that though.  (I am doing that in the new c lib with z80asm not zcc as there isn't any c source there).

The only downside at the moment is that when zcc builds libraries, the module names stored in the library are the temp filenames created by zcc during processing rather than the original source file names.

suborb, 2017-01-24:
Perhaps it should be an option instead then.

The classic library probably won't use that option - I'd prefer to use make to avoid rebuilding files that haven't changed.

I thought I'd fixed the module name thing a while back so that the right name was used.

closed 2017-01-25 by aralbrec

aralbrec, 2017-01-25:
> I thought I'd fixed the module name thing a while back so that the right name was used.

You mean by including a module directive in the asm source?  The user has to add it to asm files otherwise the temp name is used.

With sdcc compiles I think the module is pruned out if it's there right now.  There was talk at one point about having multiple module names in a file to partition individual files into separately linked pieces.  There is a case where another language is being translated to c but functions that should be linked separately are in the same source file.  The author is currently running a tool that cuts up these source files into pieces and then writes the pieces, one per file, before compiling into a library.

pauloscustodio, 2017-01-25:
The usage of multiple modules in one asm file is in the plan for z80asm2.

suborb, 2017-01-25:
> You mean by including a module directive in the asm source? The user has to add it to asm files otherwise the temp name is used.

I was talking about sccz80, but for .asm you're right. Given that zcc has the real filename, I wonder if it's worthwhile passing that as an option to z80asm so we don't need to put MODULE in the code.

commit 83505273 (2017-01-25): Fix #40 zcc_proj.lst is only created if option -z80-verb is on the compile line