z88dk/z88dk #43: Fix Up Sourceforge Page

aralbrec, 2017-01-24:
From the main menu bar I'd like to get rid of the Tickets category.  This will involve deleting the small number of old bug reports and support requests that were posted there.

I'd also like to try to delete or discontinue the commits mailing list.  If that involves deleting the mailing list is everyone ok with that?  A more thorough history is in github now anyway.  I noticed the mailing list quite often missed commits.

suborb, 2017-01-24:
I've got no issues with either of those things.

We can also do releases on GitHub so I we should additionally have the 1.99B release files here as well.

A similar question is whether we need the other mailing lists - in particular -devel only has a few subscribers, most of the traffic that used to go there is now on the forum and any discussions regarding features etc end up on these issues instead.

aralbrec, 2017-01-25:
I rearranged things bit at sourceforge.  Hid the dev mailing list and deleted the commits so that the only mailing list visible is users.  If you want to delete dev that's ok with me as long as a message is sent out first.  I think maybe we should have one mailing list at least.

closed 2017-01-25 by aralbrec