z88dk/z88dk #45: z88dk: prune unreferenced c functions from output binary

aralbrec, 2017-01-26:
This is actually in z80asm's roadmap using the module directive but the issue was originally raised in the context of c programs.

z80asm as linker extracts individual functions from a library as needed when forming the output binary.  However the unit of granularity of the extraction is the file (or rather the module, of which there can only be one per file).  So all functions and variables intended for a library, c or otherwise, should be partitioned into individual files before being added to a library to take advantage of incremental linking.

However, some tools (and people!) are generating large monolithic c or asm files containing many functions intended to behave as a library.  Unfortunately, because the unit of linking is the file, if a user program uses any function from such a large set in a single file, the linker will attach all the functions, the entire file to the output binary no matter how little of it is actually used.  Part of the reason this happens is people do not know what linking is since they are acclimatized to basic assemblers but part is due to automatic tools not partitioning their output.  Many c programmers are also often surprised to find out a function that has become obsolete or gone unused in a large program is not removed by some sort of dead code eliminator.

The proposed solution is to allow multiple .module directives in the same asm file with each .module directive delimiting the start of a new linking unit.  The c compilers would have to insert a .module directive at the start of each function.

I can think of two issues associated with this:

  * How are previously declared public and extern symbols treated in a new module?  I think the scoping directives should continue to work as they do now so that an individual module inherits all scoping ahead of its source placement.  That may mean turning any public declarations into global if the individual module is effectively processed in isolation from the rest of the source file.
  * What module name does the linker use when making object or library files?  I think it probably needs to be a combination of the filename and the module name.  If the module name is omitted then some unique (to the file) machine-generated name can be used as module name.

Some previous discussion in #40.

pauloscustodio, 2017-01-26:
The plan for z80asm has to handle each MODULE directive as a virtual source file slitter. Each module has it's own symbol table and has to include it's own EXTERN and PUBLIC declarations. I can still consider the alternative of handling these declarations at a file scope.

The module name in the linker will be composed by the source file name and the name given in the module directive.

The data model for the new z80asm is at https://github.com/z88dk/z80asm/blob/master/doc/classes.class.violet.html 

I appreciate comments.

aralbrec, 2017-01-28:
> The plan for z80asm has to handle each MODULE directive as a virtual source file slitter. Each module has it's own symbol table and has to include it's own EXTERN and PUBLIC declarations. I can still consider the alternative of handling these declarations at a file scope.

I think it has to treat these things at file scope for the c compilers.  I can see what you are proposing might also be useful at asm level so maybe there is room for both?

For the c compilers case, there are three things going on:

**1.  In post-processing we are putting the compiler intrinsics in scope by prepending declarations for the compilers' intrinsics at the beginning of the compilers' output asm.**

For zsdcc, all of this is prepended to the compiler's output:

```
1     0000              ;--------------------------------------------------------
2     0000              ; File Created by SDCC : free open source ANSI-C Compiler
3     0000              ; Version 3.6.2 #9712 (MSVC)
4     0000              ;--------------------------------------------------------
5     0000              ; Processed by Z88DK
6     0000              ;--------------------------------------------------------
7     0000              
8     0000              	EXTERN __divschar
9     0000              	EXTERN __divschar_callee
10    0000              	EXTERN __divsint
11    0000              	EXTERN __divsint_callee
12    0000              	EXTERN __divslong
13    0000              	EXTERN __divslong_callee
14    0000              	EXTERN __divslonglong
15    0000              	EXTERN __divslonglong_callee
....
....
171   0000              	EXTERN ____sdcc_store_debc_hlix
172   0000              	EXTERN ____sdcc_store_debc_mhl
173   0000              	EXTERN ____sdcc_z80_pop_ei
174   0000              	EXTERN ____sdcc_z80_pop_ei_jp
175   0000              	EXTERN ____sdcc_z80_push_di             
```

For sccz80 it is a little shorter.  Maybe that should be copied for zsdcc:

```
;* * * * *  Small-C/Plus z88dk * * * * *
;  Version: MSC.190024215 Jan 26 2017
;
;	Reconstructed for z80 Module Assembler
;	Module compile time: Sat Jan 28 10:19:12 2017

	MODULE	zzz

	INCLUDE "[z80_crt0.hdr](https://github.com/z88dk/z88dk/blob/master/lib/z80_crt0.hdr)"
```

Those are subroutines the compilers may have called in the generated code which are considered intrinsic and therefore do not appear in the compilers' own symbol tables.  These have to be visible from every function in the file.

**2. PUBLIC and EXTERN declarations are generated at file scope by the compilers.**  The compilers check that symbols are reachable for each function but they make entries for scope at the file level and output them all at once either at the beginning of the file or at the end.

For sccz80, the extern declarations from header files and any globally defined stuff in the file are scoped at the end of the source:

```
	SECTION	rodata_compiler
.i_1
	defm	"Hello world"
	defb	13

	defm	""
	defb	0

; --- Start of Static Variables ---

	SECTION	bss_compiler

._global	defs	2
	SECTION	code_compiler

; --- Start of Scope Defns ---

	EXTERN	fmemopen
	EXTERN	sprintf
	EXTERN	snprintf
....
....
	EXTERN	ftrylockfile
	EXTERN	vfprintf
	EXTERN	_fmemopen__callee
	EXTERN	fputc_callee
	EXTERN	fputs_callee
	EXTERN	putc_callee

; --- End of Scope Defns ---
```

If the main() function is to be in its own module it has to be able to see these things at the end of the file.

zsdcc is similar but puts the scoping at the top and intermingles rodata with the code after the function. 

**3. No PUBLIC/EXTERN entries are made for local variables at global scope.**

If functions are in an individual module, they have to be able to see local variables at global scope outside their module but in the same file.  I'm thinking of declarations like "static int mylocal;" at global scope.

Have to head out so I'll come back to edit this later.