z88dk/z88dk #46: zcc: module names in generated libraries are temp file names rather than original filenames

aralbrec, 2017-01-26:
zcc's -x option causes it to generate a library out of input source files.  This is a very convenient method for making libraries because zcc can receive a list of source files of any type and generate a library in a single line.

However there is one drawback :- the module names stored in the library are the temp filenames zcc creates rather than the original filenames identifiable by the user.

For example, running z80nm on one of the new c lib's libraries `z88dk/libsrc/_DEVELOPMENT/lib/sdcc_iy/zx.lib` will generate a listing like this:

```
File zx.lib at $0010: Z80RMF08
  Name: b_array_append
  Names:
    G A $0000 _b_array_append (section code_adt_b_array)
  External names:
    U         asm_b_array_append

File zx.lib at $0110: Z80RMF08
  Name: b_array_append_callee
  Names:
    G A $0000 _b_array_append_callee (section code_adt_b_array)
  External names:
    U         asm_b_array_append
.........
.........
```

The name, as in "b_array_append", is the module name or, because there are no module names in any of the new c lib's source, it is the name of the original source file "b_array_append.asm" from which that linking unit was generated.  This library was created by z80asm directly and z80asm was given the original source files to generate the library.

This library can also be built by zcc by running this from z88dk/libsrc/_DEVELOPMENT:

`zcc +zx -vn -x --lstcwd -clib=sdcc_iy -Ca--IXIY -o test @target/zx/library/zx_sdcc_iy.lst`

Running z80nm on the resulting test.lib generates this:

```
File test.lib at $0010: Z80RMF08
  Name: zcc56412
  Names:
    G A $0000 _b_array_append (section code_adt_b_array)
  External names:
    U         asm_b_array_append

File test.lib at $0110: Z80RMF08
  Name: zcc56413
  Names:
    G A $0000 _b_array_append_callee (section code_adt_b_array)
  External names:
    U         asm_b_array_append
```

The difference is the name of the module is now the temp filename generated by zcc rather than the original filename.  z80asm is being given the temp asm files after compiling/processing/whatever.  These temp filenames also show up in map files which can cause a little confusion for the user if he wonders where that function is stored.

zcc knows the original filenames of these files so I wonder if there is a way to communicate that to z80asm.

suborb mentioned:
> I was talking about sccz80, but for .asm you're right. Given that zcc has the real filename, I wonder if it's worthwhile passing that as an option to z80asm so we don't need to put MODULE in the code.

zcc is going to be passing a mass of files not usually one at a time and quite often as a list of files in a single .lst file to avoid command line limitations.  So I don't think there is any way to add a command line parameter sensibly.

What about a new .file directive?  zcc could add it automatically at the beginning of a .asm file before it assembles.  If it's not present, use the filename.

I'm also thinking a little ahead in terms of multiple modules in the same .asm file #45 where the name of the linking unit inside a file might be filename/modulename

Previous discussion in #40

aralbrec, 2017-01-26:
> zcc is going to be passing a mass of files not usually one at a time and quite often as a list of files in a single .lst file to avoid command line limitations. So I don't think there is any way to add a command line parameter sensibly.

Actually that's not right.  I couldn't remember if zcc passes files at the .asm stage or the .o stage to z80asm for building a library but it is the .o stage.  So that means zcc individually assembles each .asm file and it would be possible to pass the original filename as command line parameter to z80asm.

Right now that parameter would become the module name.

pauloscustodio, 2017-01-26:
z80asm MODULE directive does just that. It replaces the source file name by the name given as argument in the linked file. zcc just has to add "MODULE original_file" to every temporary file before assembling.

In my plan for z80asm2 the linked module will be named <source_file>_<module_directive_name>

aralbrec, 2017-01-26:
Will z80asm be okay with receiving two module directives?

If zcc inserts one in the front and the user has specified one which would appear second, will z80asm be able to use the last one declared?

> In my plan for z80asm2 the linked module will be named <source_file>_<module_directive_name>

Yes this is what I was thinking about.  In the future, it's the source_file name that zcc should be tweaking and not the module.

pauloscustodio, 2017-01-26:
> If zcc inserts one in the front and the user has specified one which would appear second, will z80asm be > able to use the last one declared?

Today there's an error message on the second module directive. But this can be easily changed, as the module name is not used except for the name in the .o file.

closed 2017-01-26 by aralbrec

aralbrec, 2017-01-26:
zcc now inserts a module directive at the start of every file at the .asm stage.

  * We've moved away from shell invokes of copy and cp to shell invokes of type and cat for appending
  * CRTs must have module directives at the top since they are processed outside zcc's loop

To form the module sting, the basename is taken and non-alnum chars are replaced with '_'.  So a file "clisp.c" will get module name "clisp_c" and a file "clisp.asm" will get module name "clisp_asm".

Tested on windows and this change fixes the zcc lib issue as well as makes the map file clearer.

commit 124a0ed7 (2017-01-26): Fix #46 zcc inserts MODULE directive into every file processed at the .asm stage