z88dk/z88dk #49: z80asm: z180 has been removed?

aralbrec, 2017-01-30:
Has the z180 as assemble target been removed?

If so is it possible to add it back?  There are only a handful of instructions different from the z80:

**New Instructions Operation**
  * SLP Enter SLEEP mode
  * MLT 8-bit multiply with 16-bit result
  * INO g, (m) Input contents of immediate I/O address
  * OUT0 (m), g Output register contents to immediate I/O address
  * OTIM Block output - increment
  * OTIMR Block output - increment and repeat
  * OTDM Block output - decrement
  * OTDMR Block output - decrement and repeat
  * TSTIO m Non-destructive AND, I/O port, and accumulator
  * TST g Non-destructive AND, register, and accumulator
  * TST m Non-destructive AND, immediate data, and accumulator
  * TST (HL) Non-destructive AND, memory data, and accumulator

[z180 cpu user manual](https://github.com/z88dk/techdocs/blob/master/zilog/z180_cpu_um50.pdf)

pauloscustodio, 2017-01-30:
As far as I know, there was never a z180 support in z80asm. There is a partial support for the Rabbit (different names for I and R register and CALL flag emulated by JR !flag; CALL).

I can add z180 as a target in the road map.

pauloscustodio, 2017-01-30:
Z180 support has been added with the new --cpu=z180 option.

commit a58f42a1 (2017-01-30): Fix #49 implement z180 as a target

Add --cpu=z180 option to allow the additional Z180 opcodes:
- SLP Enter SLEEP mode
- MLT 8-bit multiply with 16-bit result
- INO g, (m) Input contents of immediate I/O address
- OUT0 (m), g Output register contents to immediate I/O address
- OTIM Block output - increment
- OTIMR Block output - increment and repeat
- OTDM Block output - decrement
- OTDMR Block output - decrement and repeat
- TSTIO m Non-destructive AND, I/O port, and accumulator
- TST g Non-destructive AND, register, and accumulator
- TST m Non-destructive AND, immediate data, and accumulator [... cut]
closed 2017-01-30 by pauloscustodio