z88dk/z88dk #50: Z180 target development - enhancement

feilipu, 2017-01-30:
I would like to request an enhancement to support an embedded z180 target.

I'm working on a [Z180 platform](https://feilipu.me/2016/05/23/another-z80-project/) currently, and there is also an option to add a Z180 processor to the [RC2014 platform](http://rc2014.co.uk/) with little effort.

The z180 has a number of additional machine instructions that are mentioned in issue #49.

Additionally, major differences to the z80 include the ability to flexibly assign the location of the z180 I/O base vectors, and its internal interrupt vectors. These are referred to as [IO_BASE and VECTOR_BASE in an example initialisation file](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/yaz180_NascomBasic56k/int56k.asm).

The IO_BASE is loaded into register ICR, and VECTOR_BASE is loaded into register IL, to determine the initial configuration of the z180. These can be redefined as required following initialisation.

Further other initialisation option examples are found in the INIT: section for reference.

aralbrec, 2017-01-30:
I'll take that up for the new c library.

On first pass we can assume a 64k address space so that the memory map will be set up by the crt similar to the z80 by default and we can add the user-definable IO_BASE and VECTOR_BASE initialization in the crt startup.

The new MLT instruction adds the possibility of increasing the speed of the integer math routines so we can have a look at adding alternate z180 implementations of those.

sdcc can be put in a z180 mode - the difference seems to be the occasional generation of MLT and maybe differences in the sfrs - but we'd have to keep it in z80 mode until the assembler supports the extra z180 instructions.

We'll have to check if sdcc's peepholer can properly deal with the z180 instructions.

aralbrec, 2017-01-31:
A new +z180 target was added to the new c lib.  This target is a copy of the [generic z80 target (+embedded)](https://www.z88dk.org/wiki/doku.php?id=libnew:target_embedded) so the same documentation applies.

The difference is that the z180 features of zsdcc and z80asm are enabled during compiles.  zsdcc is not accepting the "--portmode=180" command line flag so this will have to be looked at later.  That flag is supposed to enable generation of IN0 / OUT0 instructions for 8-bit port access in zsdcc.

This also means that z180 mnemonics should be accepted in .asm files added to a project.

What we still need to look at:

  * Ensure zsdcc correctly understands z180 instructions in the peepholer.  If not this impacts negatively on optimization.
  * Replace the standard z80 startup code with z180 specific startup code that initializes the z180's on-chip hardware.
  * Go through the C library and specialize for the z180 where applicable.

feilipu, 2017-01-31:
Great to see progress on the z180.

I'm very keen to dive into using and testing this work on my [z180](https://feilipu.me/2016/05/23/another-z80-project/). But, at the moment I'm struggling with implementing a USB based [bootloader-free in-circuit parallel programming (ICPP) solution](http://www.ecstaticlyrics.com/electronics/Z80/EEPROM_programmer/) for the board. The EEPROM PLCC socket used on my platform has a short lifetime (I guess about 100 cycles), and therefore I hesitate to continually re-seat the Flash during development process.

As soon as I've progress on the ICPP front, I'll dig into testing and helping get this further along.

For interest, are there any other z180 platforms in easy reach?

aralbrec, 2017-02-01:
> For interest, are there any other z180 platforms in easy reach?

I'm not aware of any to be honest.  I think most z180 machines back in the day were business or technical targets so didn't gather much of a following among common users.

There's no rush -- there is still work to do on the target.

Remaining:

  * Replace the standard z80 startup code with z180 specific startup code that initializes the z180's on-chip hardware.
  * Go through the C library and specialize for the z180 where applicable.

aralbrec, 2017-02-01:
@suborb 

zsdcc has to be re-built to gather the changes for the z180.
Patch as usual:  https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_z88dk_patch.zip
Windows binaries are in that zip.  I've updated the windows build to the latest sdcc 9833.

suborb, 2017-02-01:
Done, the build tonight will have them.

aralbrec, 2017-02-22:
Nearing completion of the Z180 target:

  * The Z180 target can be configured as Z80180 class or Z8S180 / Z8L180 class.  The difference is the latter has more internal i/o registers.
  * #defines (sccz80) and sfrs (sdcc) are automatically generated for the Z180's internal peripherals.  These are available from C by including <arch.h> (although the preferred include will be <z180.h> soon) and are always available from asm using zilog's names and accompanying EXTERN.  The __IO_BASE_ADDRESS setting in config_cpu.m4 determines the i/o base address of internal registers and the adjusted port address is reflected in the headers.
  * The interrupt vector base is configured with __IO_VECTOR_BASE in the same config file.
  * The crt will initialize both IL and ICR with __IO_BASE_ADDRESS and __IO_VECTOR_BASE.
  * By default the interrupt vector table is automatically generated by the crt in section "code_vector".  The table is initially zero size and it is up to the user to create the contents of the vector table.  An example might have the vector table defined in a separate file "interrupt.asm":
```
SECTION code_vector

; vector 0
EXTERN _isr_clock
defw _isr_clock

; vector 2
EXTERN _isr_pio
defw _isr_pio

.. etc
```
The crt-generated vector table will be located at 0x100.  Adding "interrupt.asm" to the compile line will place the table above at address 0x100.  The crt will automatically intialize the I register with the vector table address.  This behaviour can be modified by changing the value of the "CRT_ORG_VECTOR_TABLE" pragma.  The default behaviour is achieved with a value of -1.  Changing to any positive address will place the vector table at that address (lower 8 bits zeroed).  A compile will generate the table in a separate binary "foo_code_vector.bin" which you must find a way to get into the right memory location.  The crt will again initialize the I register correctly.  Getting the assembled table to the right address can be done using two successive compiles.
* The crt now places code in a "COMMON0" section rather than a "CODE" section.  This better matches the memory spaces defined by Zilog.  The data and bss sections are separately generated as per the z80 embedded target.  User programs can create bankswitched programs manually.

Still to do:

  * Generate z180.h to replace z80.h.
  * Specialize the library for the z180 where necessary.  The exact T-state delay function in z80.h will not work properly and the integer math functions can be improved.
  * Do a better job of keeping system-defined symbols out of the user's symbol space.

aralbrec, 2017-02-23:
It appears it is not possible to handle TRAPs unless the IO_BASE_ADDRESS is 0 or self-modifying code is resorted to.

The z180 executes a trap if it encounters an illegal instruction.  A trap event causes a jp to address 0 which means address 0 doubles as power on reset and trap.   To distinguish the two, you are supposed to read the ITC register and check if bit 7 is set (trap) or not (power on reset).

But what port is the ITC register mapped to?  At power up reset it will be 0x34.  If a trap is occurring it will be 0x34+IO_BASE_ADDRESS.  It's impossible to know which port is the correct one to use at address 0.  The only way around this is to have IO_BASE_ADDRESS=0.

If code at address 0 is running out of ram, it would be possible to patch the code at 0x0 after power on initialization to use IO_BASE_ADDRESS+0x34 to test for a TRAP.  But this requires an all ram system which is not the most common type.

I think the only reasonable solution is to forbid the use of i/o address 0x3f except for the internal ICR register even if the z180 io base is offset away from 0.  Then at address 0, we can do "out0 (0x3f),0" to set the z180's registers to a known location, test the ITC register for trap and restore io_base to its runtime value.

Am I missing something?  Otherwise this is what I think we have to do.

aralbrec, 2017-02-23:
> Then at address 0, we can do "out0 (0x3f),0" to set the z180's registers to a known location

Ha, can't do that because we don't know where the ICR register is either.  Maybe it has to be:

```
push af

xor a
out0 (IO_BASE_ADDRESS+0x3f),a   ;; move iobase back to zero

in0 a,(0x34)   ;; find out if it's a trap
bit 7,a
jp nz, __TRAP

; continue with power on initialization

ld a,IO_BASE_ADDRESS
out0 (0x3f),a   ;; move io base to expected location
```

Doing the above means io addresses 0x3f and 0x34 are not allowed for the hardware design.

feilipu, 2017-02-23:
Z180 in(c) and out(c),are 16 bit capable, so that's a reasonable caveat to impose.

It's only one amateur design, but I've kept physical I/O right out of the page 0 address space. So there couldn't be conflicts. I haven't seen other solutions in detail. 

Are we ok doing a push with the SP in an unknown state? I guess worst case it is writing to 0xFFFF.

aralbrec, 2017-02-24:
> From the datasheet, page 44...
> >  Normally, external I/O addresses should be chosen to avoid overlap with internal I/O addresses and duplicate I/O accesses.

At the very minimum, no external hardware should be mapped to 0x003f because the startup code has to write to that internal register to change the IO_BASE_ADDRESS.  With that requirement, I think this solution will work:

```
; address = 0x0000
; must decide if this is a power on reset or a trap

   push af
   
   ; move i/o base address back to 0
   ; harmless on power up when ICR is not mapped yet
   ; note that this resets the IOSTOP bit
   
   xor a
   out0 (ICR),a

   jp __Test_Trap
   
; address = 0x0008
....
....
__Test_Trap:

   ld a,__IO_BASE_ADDRESS
   out0 (0x3f),a

IF __crt_enable_trap

   in0 a,(ITC)
   and $80

   jr z, __Start
   
   pop af
   
   EXTERN _z180_trap
   jp     _z180_trap

ENDIF

;; fall through to __Start

```

A zero is written to port  (0x3f + IO_BASE_ADDRESS) to move the internal registers back to base 0.  On power up reset, that should be a do-nothing write since the base address starts at 0.

Then we restore the IO_BASE_ADDRESS by writing to port 0x3f.  Now the internal registers are known to be mapped at their final locations.  Bit 7 of port ITC is checked for TRAP and a decision is made on whether to serve the trap or to continue with initialization.

The only thing is does it matter that IOSTOP is set to 0 when a TRAP happens?

>Are we ok doing a push with the SP in an unknown state? I guess worst case it is writing to 0xFFFF.

The Z180 sets SP=0 at power up so I think that's ok.

aralbrec, 2017-02-24:
Actually I think I may have been spinning in circles.

This should be enough:

```
; address = 0x0000
; must decide if this is a power on reset or a trap

   push af
   
   ; move i/o base address back to 0
   ; harmless on power up when ICR is not mapped yet
   
   ld a,__IO_BASE_ADDRESS
   out0 (0x3f),a

   jp __Test_Trap
...
...
__Test_Trap:

IF __crt_enable_trap

   in0 a,(ITC)
   and $80

   jr z, __Start
   
   pop af
   
   EXTERN _z180_trap
   jp     _z180_trap

ENDIF
```

If the IO_BASE_ADDRESS has already been set, then that first "out0 (0x3f),a" will do nothing.  However if it's power on reset, that out0 will set the correct IO_BASE_ADDRESS.  The bonus is there is no longer a problem with the IOSTOP bit.

I think this is what Zilog had in mind.

feilipu, 2017-02-24:
Yes, independently this is exactly the ordering that I have followed, except I ignored (for simplicity) the `TRAP`.
First thing in my [`INIT`](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/yaz180_NascomBasic32k_HexLoadr/int32k.asm), set the `IO_BASE`. Second thing, set the `VECTOR_BASE` outside the `RST` and `NMI` address space.

``` asm
;------------------------------------------------------------------------------
INIT:
                                         ; Set I/O Control Reg (ICR)
               LD        A,IO_BASE       ; ICR = $00 [xx00 0000] for I/O Registers at $00 - $3F
               OUT0      (ICR),A         ; Standard I/O Mapping (0 Enabled)

                                         ; Set interrupt vector base (IL)
               LD        A,VECTOR_BASE   ; IL = $80 [001x xxxx] for Vectors at $80 - $90
               OUT0      (IL),A          ; Output to the Interrupt Vector Low reg
 
```

aralbrec, 2017-02-24:
Yes it looks like we followed similar paths.

I've also copied your 0x80 address vector table as an option for the z80 in im2 mode and as the default for the z180.  This requires I=0 and interrupt vector ids being assigned at least 0x80 (IO_BASE_ADDRESS=0x80) so that code can exist up to address 0x80.

The option is chosen with CRT_ORG_VECTOR_TABLE=-0x80 and the regular option with full size im2 table at 0x100 is chosen with CRT_ORG_VECTOR_TABLE=-0x100.  These are negative to tell the crt to make those tables inside the page zero code.  Positive values generate a table at some address in memory under programmer responsibility.

The [https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/crt_page_zero_z180.inc](final page zero code for the z180) looks a little messy but that's because I am trying to fill the gaps with common library routines.  It comes down to this for power up / trap:

```
; address = 0x0000
; must decide if this is a power on reset or a trap

   push af
   ld a,__IO_BASE_ADDRESS

   jp __Test_Trap
   
; address = 0x0008
...
...
__Test_Trap:

   out0 (0x3f),a

IF __crt_enable_trap

   in0 a,(ITC)
   and $80

   jr z, __Test_Trap_Fail
   
   pop af
   
   EXTERN _z180_trap
   jp     _z180_trap

ENDIF

__Test_Trap_Fail:

   include "../crt_start_di.inc"

   IF REGISTER_SP = -1
   
   pop af
   
   ENDIF
   
   ld a,__IO_VECTOR_BASE
   out0 (IL),a
   
   im 2

;; fall through to __Start
```

I decided to set im2 mode for the z180 too; this affects /INT0 and z80 peripherals.  I am still waffling on this since the z80 crt defaults to im1 but with the vector table always there in the z180 maybe im2 is a more common choice.

feilipu, 2017-02-24:
> With the vector table always there in the z180 maybe IM2 is a more common choice.

Just for reference, for simplicity I decided to use the static IM1 mode for the Z80 compatible interrupt `INT0`.

The Z180 interrupts `INT1` and `INT2` also use an equivalent static vector of IM1 type, although they're in a vector table that can be relocated.

For Z180 from the datasheet...

> During the `INT0` Mode 2 acknowledge cycle, the low-order 8 bits of the vector is fetched from the data bus at the rising edge of T3 and the CPU acquires the 16-bit vector.

Using the `IM1` option, I don't have to give the peripheral any intelligence to fill in the vector. It simply has to toggle an interrupt line. This does limit the total number of external interrupts to 4 (including the `NMI`) but I felt that was sufficient for my (personal) purposes. Particularly, because the Z180 wraps the ASCI, PRT, DMA, and other system controllers into internal interrupts, which in the Z80 world would have to be accommodated by external interrupts.

aralbrec, 2017-02-24:
>Just for reference, for simplicity I decided to use the static IM1 mode for the Z80 compatible interrupt INT0.
>
>The Z180 interrupts INT1 and INT2 also use an equivalent static vector of IM1 type, although they're in a vector table that can be relocated.

I'm not sure what you mean by static vectors of IM1 type for /INT1 and /INT2?

> NMI, INT1, and INT2 feature fixed interrupt response modes. INT0 has 3
> different software programmable interrupt response modes—Mode 0,
> Mode 1 and Mode 2.

Unless you have pull ups / pull downs on the external data bus to provide a default vector id?

But I can see the use of this.  I think I will just go ahead and add another crt variable for convenience to optionally set a chosen im mode.

feilipu, 2017-02-24:
> I'm not sure what you mean by static vectors of IM1 type for /INT1 and /INT2?

To explain, from the datasheet...

> INT1, INT2
> The operation of external interrupts INT1 and INT2 is a vector mode **similar** to INT0 Mode 2. The difference is that INT1 and INT2 generate the low-order byte of vector table address using the IL (Interrupt Vector Low) register **rather** than fetching it from the data bus.

`INT1`, and `INT2` operate differently in the Z180 to in the Z80. In the Z180 they are linked to a specific low byte address, and the final vector not taken from the data bus. So they are static vectors assigned to a specific value within the `VECTOR_BASE` range.

So it is only `INT0` that has the capability to generate a flexible vector, depending on what address the interrupting peripheral puts on the data bus, and only `INT0` can operate in a chained mode because of this.

feilipu, 2017-04-18:
If it was possible to include some ASCI drivers for the Z180 before the release #153, then I can provide some source to enable that.

I would have liked to have drivers for all the Z180 peripherals ready by now, but since a release is due and there are a few Z180 boards from different sources due out shortly, I'd like to go with what we have to get the ASCI and memory models up and running.

I am fairly happy with the generic ASCI driver code that I've been testing on my [YAZ180](https://feilipu.me/2016/05/23/another-z80-project/) over the past months, and I think it is stable enough to be optimised into the z88dk way of doing things. I'll reference parts of the code here for discussion.

Ideally, I would make a pull request, but the "right" outcome for z88dk is probably best done by someone who recognises the right way to dis-aggregate the code into the right M4 incantations.

Note: the copyright text on my files is 'left over' from Grant Searle's headers. Please take this as permission / advice that the notice can be removed, and the z88dk standard licence applied.

## The header files.
 [yaz180.h](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/yaz180.h) includes a bunch of Z180 specific information (internal registers, and their configuration bits), together with definitions for the Z80 and Z180 vector tables.

The ASCI code depends on having 256 byte buffers for both transmit and receive ASCI code, and that these [buffers are located on the low byte zero](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/yaz180.h#L404) boundary. I have provided buffer definitions for ASCI0 and ASCI1. Probably, it would be best to add other peripheral buffers after the ASCI provisions (rather than before, as I've inappropriately done).

Other variables relevant to the ASCI drivers are located after the Z80 and Z180 vector tables, and before the buffers which occur at 0xNN00.

The Z80 vector (jump) table can be located anywhere.
The Z180 vector (address) table has to be located on a 32 byte boundary.

The Z80 vector prototype is written at 0x0040 (between INT0 and NMI vectors).
The Z180 vector prototype  is written at 0x00C0 (space from NMI is used for TRAP vector code in this case).

[z80intr.asm](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/z80intr.asm) sets up the Z80 RST, INT0, and NMI interrupts and jump table prototype.
As noted the prototype jump table is located from 0x0040.
There are the null return instructions RETN, EI RETI, and RET located from 0x0060 as well.

[z180intr.asm](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/z180intr.asm) sets up the Z80 internal interrupts, the INT1 and INT2, and the TRAP / RESET code (sourced from z88dk).

It is left to the user's code to define what is included in each of the Z80 and Z180 interrupt addresses.
An [example is provided](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/yaz180_NascomBasic32k_HexLoadr/int32k.asm#L418) here. I'm sure that this could be better handled by weak defaulting and only changing the definition if the user provided an alternative routine.

## The ASCI Code
The ASCI code provides an[ interrupt driven receive and transmit](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/yaz180_NascomBasic32k_HexLoadr/int32k.asm#L46) of bytes. Reception makes use of the 4 byte hardware buffer present on the Z8S180, but this is not present on other Z180 versions. Receive errors lead to discarded bytes.

The [Rx code](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/yaz180_NascomBasic32k_HexLoadr/int32k.asm#L118) will wait until a byte is available to present, if none are available.

The [Tx code](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/yaz180_NascomBasic32k_HexLoadr/int32k.asm#L138) is optimised to first check whether a byte can be directly transmitted (achieving polled Tx latency), before otherwise poking it into the Tx buffer. If the Tx buffer is full, it will busy wait (like a polled Tx function).

It would make sense to provide these functions for both ASCI0 and ASCI1 interfaces. If inclusion of one, or both could be configurable it would be most optimal.

## Z180 setup code.

I've written some [Z180 setup code](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/z180intr.asm#L27) that could be included as part of the TRAP routine. This code locates the I/O registers, disables all interfaces, prior to the user doing something more appropriate, and then builds the vector tables in RAM. I think this code could be considered generic enough to be included in any Z180 hardware initialisation.

There is also some more [specific Z180 setup](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/yaz180_NascomBasic32k_HexLoadr/int32k.asm#L308) that configures the memory model, and initialises the ASCI devices. The ASCI initialisation code should be generalised to be passed in a baud rate & signal type, and generate a configuration based on this info. I haven't done this work yet (sorry).

aralbrec, 2017-04-19:
Yes I can have a look at this when I get back to the z180.  The library build for the z180 can specify what type of z180 device it is (z80180, z8s180, z8l180) so it's possible to conditionally include code for the 4-byte buffer in the ASCI when supported.  The m4 instantiations for the ASCI can insert start up code that initializes baud rate, etc.

There is a near-complete concept of non-blocking i/o in the stdio library so I will have to see if this can do non-blocking i/o for the ASCI.  If that's the case I can probably turn that into a feature bit in the drivers.

The other concern I have is the proliferation of symbols in the global namespace.  There are already a lot defined by Zilog and I've kept them as-is because they are well known names but if they are not manufacturer-defined I'd like to namespace constants with leading two underscores and a device name if applicable.  Header files can expose the internal names into the global namespace when included (`#define THIS_THAT  __THIS_THAT`).

feilipu, 2017-04-19:
> The other concern I have is the proliferation of symbols in the global namespace.
I think the only variable labels that are relevant for the global namespace are those required to write an ISR address into the active (RAM) Z80 or Z180 vector table at run time. Everything else can be safely hidden.

feilipu, 2017-05-18:
It is not yet substantially tested (I'll need some more time for testing), but I've now a [driver for IDE](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/yaz180_UnitTest/z180ide.asm) from the Z180 platform using the 82C55 PIO. Depending on the timeline to close of Release 1.99c, I would like to either:
1. modify the driver to integrate with the [FatFS code from Elm ChaN](http://elm-chan.org/fsw/ff/00index_e.html). I've used this code often, and it is very fully featured (and well supported).
2. modify the driver to work with an existing z88dk FatFS system.
3. insert it as is with appropriate z88dk sanitation, and then use glue functions to connect to a FatFS system. This will be the option if R1.99c is closing imminently.

Any guidance welcome, please.

aralbrec, 2017-05-19:
In the newlib the disk portion of stdio is not complete yet.  This is the next major topic in my to-do list and I wasn't planning to tackle it until after 1.99c.  Completing this would be a milestone so I'd expect another release after it's done.  In other words, it's probably best to have disk i/o properly integrate after 1.99c.

The implementation is currently missing fleshed out file and directory management like fopen and fclose and mounting points.  The latter would connect a directory hierarchy to drivers that would take care of file and directory management.  It is possible to do stream i/o to disk devices.  A simple implementation would have the current library distill read/write requests to getchar and putchar which a disk driver could use to manage cached sectors and read/write bytes using file pointers stored in the FILE*/fd structures.  The ultimate implementation will be taking care of sector caching and will forward logical sector read/writes to the driver.  FATFS can insert itself between a low level disk device driver and stdio to turn these logical sector read/writes into physical ones.

So we don't have FATFS yet and the one you linked is one I've been looking at as a guide to introducing one.  If you want to get at the z88dk stream i/o instead of using a separate one from elm-char I can probably help with that as a temporary measure until the rest is ready but I think this will also take a little effort to do.

1.99c I'd like to see out the door very soon.

feilipu, 2017-08-12:
Given the status with #312 z80-zxn and rabbit with additional h/w multiply capabilities, it would be good to look at how the z180, r2k, r3k, zxn fpga multiply instructions can be added to newlib math.

I've been hunting for generic integer n bit integer and float math routines, based off an underlying 8x8 or 16x16 h/w multiply, but I've come up short. But, I'm sure there must be something out there, because of the age of the h/w.

Thoughts?

aralbrec, 2017-08-12:
I've never seen anything using specific hw multiply instructions, except suborb linked digi's rabbit repo on github a while back.  They have some source code in there.

The integer math should be fairly easy to do anyway, just have to sit down and do it.  I'm not sure if the mult instructions can be used in division.

feilipu, 2017-08-12:
So a bit more search-fu has turned up some references based on the PIC17, which has an 8x8 multiplier.
In this MicroChip document, [Page 67 has the basic signed and unsigned 8x8 and 16x16 algorithms](https://github.com/z88dk/z88dk/files/1219908/30289b.pdf).

There's also a lot of routines outlined in [this document](https://github.com/z88dk/z88dk/files/1219909/00575.pdf).
But the translation from PIC17 mnemonics would not be straightforward.

Perhaps that helps?

feilipu, 2018-03-07:
From my point of view this is done done, and can be closed.

The remaining related open issue is my need to have the memory models between [Z80](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/crt_memory_model_z80.inc#L15) and [Z180](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/crt_memory_model_z180.inc#L18) different, to support the need to have a [YAZ180 specific preamble](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/yaz180/crt_preamble.asm) code after the standard vector code, but before the standard init code.

I guess this should be resolved to a single memory model, so a trap is not being set for later.

closed 2018-03-07 by feilipu