z88dk/z88dk #51: new c lib: scanf of floats %aefg unsupported

aralbrec, 2017-01-30:
In the new c lib you can printf floats but you can't scanf them.  The recommended workaround is to first scan as a string using %s or %[ and then use strtod(), atof() or one of the non-standard dto?() functions to do the float conversion.

What is missing is a state machine in the stdio library that examines an incoming char stream and decides where a float ends.  This is complicated by the fact that floats can take on a number of forms:

  * 123
  * 123.
  * .
  * .123
  * 123.123
  * variations with exponent e
  * hex floats 0xabc.123P-28
  * special strings "nan", "infinity" etc

suborb, 2017-01-30:
There's code that handles the common cases in the classic library - we read from the file and put it into a temporary buffer on the stack which atof() then parses. It's in libsrc/stdio/__scanf_handle_f.asm 

It follows the 99% rule so would need extension for the new library, but could be a quick win whilst you work on full c99 compatibility.

aralbrec, 2017-01-30:
Yes maybe a temporary partial solution is ok.  The underlying implementation of strtod/atof in the new c lib can do all the float formats listed above so it just needs a buffer to act on that is gathered by scanf.  But maybe a %f only solution can be implemented for the decimal float case so that a simple state machine can be employed.