z88dk/z88dk #52: new c lib: need a way to communicate target and library configuration to the c program

aralbrec, 2017-01-30:
This somewhat applies to the classic lib too.

The mechanism for including target-specific information in the new c lib is to `#include <arch.h>` where that information should be made available via defines.  Inside arch.h there should/will be a bunch of #ifdefs guarding info for each specific target with the #ifdef argument coming from the target.cfg file.  So, eg, zx.cfg will add -D__SPECTRUM to all compiles so that arch.h can do `#ifdef __SPECTRUM .... #endif `

However the target and library configuration is currently only available from assembler in z88dk/libsrc/_DEVELOPMENT/target/$ARCH/clib_cfg.asm and clib_target_cfg.asm and there is no way to communicate that information to the c program.  Some things don't need to be communicated but some things should be like cpu clock rate, number of display rows managed by a sprite engine, etc.

I can see two ways to do this communication:

  1. Create a c header file in /_DEVELOPMENT/target/$ARCH/.. that is manually kept consistent with the two cfg files by the user.  Then arch.h can be made to include this header file.
  2. Make the assembly defines in the cfg files public so that they are accessible to user c and asm subroutines.  The c program (through arch.h) can access constant values defined in asm using an array syntax:

```
Feature defined in cfg file:

PUBLIC _FEATURE
defc _FEATURE = 0x1234

Feature accessed in arch.h:

extern unsigned char FEATURE[];
```

The address of the "FEATURE" array will equal the FEATURE value defined in asm.

The advantage of the second method is that everything is kept consistent automatically although it is a little bizarre.

suborb, 2017-01-30:
The second method, although bizarre, is my preference. The classic library tends to tune things through the CRT rather than the library configuration, so it needs to be automatic rather than manually kept in sync.

aralbrec, 2017-01-30:
How the configuration is handled in the new c lib is a bit muddied because at the time it was not clear what was required.

It seems there has to be a distinction made between what needs to be known at build time and what needs to be known at runtime with some info in both categories.  The build time stuff has to be included in the source file so that conditional assembly can take place as the library is built (ie no PUBLIC).  The runtime stuff has to be assembled as part of every compile and exported via PUBLIC.

aralbrec, 2017-02-02:
Ok I think I have a good method for this.

The [first m4 source file](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/arch/zx/nirvanap/z80/asm_NIRVANAP.asm.m4)  in now in z88dk/master for the new c lib but I think I will redo all the fast integer math routines using m4 too when the z180 versions are committed.  The fast int math routines are hard to read because there is a lot of unrolling; changing to m4 will allow loops to replace the unrolling.

So the library build process needs an m4 step and I will add that to all the new c lib targets.  The m4 step will run before z80asm builds the library and will use "zcc -m4" to generate the associated asm file just after the library config files have been copied.  It will be fed a .lst of .m4 macros to process that will be stored along with each target's list of library files.

The good thing about this is we can also use this m4 step to generate c headers and asm files based on the configuration information.

So from assembler we'd define one configuration in a file without PUBLIC.  This file is included by many library source files to perform conditional compilation.

A second file would be generated by m4 that would include that configuration file but add PUBLIC.  This file would be included into the crt when the program is compiled and would make this configuration information available to the linker at program compile time.

Another header file would be generated by m4 in the target's directory that create #defines for all the configuration information.  This header file would be included into arch.h and would be accessible by C programs by including <arch.h>

Making the config info available via #define grants access to the preprocessor for conditional compiles.  The array method is a runtime value only.

aralbrec, 2017-02-16:
> The second method, although bizarre, is my preference. The classic library tends to tune things through the CRT rather than the library configuration, so it needs to be automatic rather than manually kept in sync.

It turns out that the compile time pragmas cannot be easily turned into m4 defines so those I have left as is in the new c lib.

To export FOPEN_MAX and OPEN_MAX, I'm using the array address kludge as well:

```
extern unsigned char    _clib_fopen_max[];
#define FOPEN_MAX       ((unsigned char)(_clib_fopen_max))
```

This also means these are a runtime values only.

aralbrec, 2017-02-18:
The z180Target branch contains a renovation of the new c lib for communicating configuration information to both C and asm.  After some more testing, I'll be merging it into master.

I think this is a very good way to do things.  I'll use the zx target as example since it has the most complicated configuration of all the targets.

**target/zx/config.m4**
This is the master configuration file that aggregates individual config files.  The configuration is spread across logically separate files so that the user can easily find the options that he may want to edit.

For the zx target these are the individual config files:  config_target.m4, config_cpu.m4, config_clib.m4, config_clib_const.m4, config_sound_bit.m4, config_bifrost_2.m4, config_bifrost_h.m4, config_biforst_l.m4, config_nirvanam.m4, config_nirvanap.m4, config_sp1.m4.

**target/zx/config_target.m4**
Supplies information on specific spectrum model and defines for testing in user code.

**target/zx/config_cpu.m4**
Information on the cpu (z80 in this case) including whether it is cmos or nmos, clock rate and what, if any, unreliable instructions are allowed in the library.

**target/zx/config_clib.m4**
The is is the c library configuration that allows a wide range of choices affecting performance and code size.

**target/zx/config_clib_const.m4**
These define constants like error numbers, stdio message identifiers, ioctl identifiers, etc and for the target can include i/o or memory mapped hardware addresses.

All these files are m4 files that the user can edit.  The file is divided into four parts:  the first contains the user-editable area with m4 defines.  The second is an area that generates PUBLIC for export of symbols at compile time.  The third contains symbol definitions in asm (defc) for conditional assembly at library build time and the last contains #define for C.

When the library is built, the first step taken is to process the target's config.m4 in order to generate three files:  config_TARGET.h, config_TARGET_private.inc and config_TARGET_public.inc.

**config_TARGET.h** contains all the symbols in C preprocessor #define form.
**config_TARGET_private.inc** contains all the symbols in asm defc form.
**config_TARGET_public.inc** contains all the symbols in asm defc form and exported with PUBLIC.

The file config_TARGET_public.inc is included in the crt at compile time so that is how the configuration information is available to assembly language during the compile.

The file config_TARGET_private.inc is used while the library builds.   The build script copies the specific target's file to config_private.inc in the build directory and then the library source files refer to it as needed for build-time configuration.  This enables conditional assembly.

The file config_TARGET.h sits in the target's home directory.  A "#include <arch.h>" will include that header - the selection is made via the target's cfg file where a small number of c preprocessor defines identify what target is compiling.  So, eg, for the zx target "__SPECTRUM" is defined and for the z180 target "__EMBEDDED_Z180" is defined.  <arch.h> just acts like a switch, including the correct target's config_TARGET.h file from its home directory.  This is how configuration information is communicated to the C program.

There are some interesting things that can be done with this:

* For hardware devices that can be mapped to different i/o addresses, the compile time configuration keeps C header up to date automatically.  An example is the z180 target (this part is not complete yet) where the base i/o address if built-in peripherals is user definable.  The configuration can select the base address and the C header will automatically reflect the correct i/o addresses for peripherals including defining sfrs for sdcc.

* Separation of user and system symbols can be maintained.  If the system symbols always have a leading "__" the C headers can control what symbols are exported into the user space with suitable defines in headers as in "#define EOK __EOK".

* The new c lib now has a z88dk version variable available (1992 corresponding to 1.99C) that is changed in one place.

* Some defined cpp symbols can be redefined for more information.  For example, for all zx compiles, the zx.cfg file ensures that "__SPECTRUM" is defined but including <arch.h> will change that value to indicate which spectrum model is being compiled for.

For the new c lib, a code inspection needs to be done to achieve the system/user symbols separation which I think is a very worthwhile thing to do.

commit f93b5ddc (2017-02-19): Fix #52 and #69 with new method for target configuration in the new c library
closed 2017-02-19 by aralbrec