z88dk/z88dk #58: Locate a variable at a specific (general) RAM address - Question

feilipu, 2017-02-05:
I'm interested to know whether it is possible to allocate a piece of memory (for a ring buffer) that falls exactly on the low byte address boundary of RAM. Essentially the upper address byte is "don't care". But to simplify bounds checking, it is convenient to have a ring buffer start on 0xnn00 address.

I'd rather not have to assign a fixed starting address. That is not general enough, and is difficult to maintain.

Specifically, I can reduce an [interrupt time substantially](https://groups.google.com/d/msg/rc2014-z80/hlhxDpDDnR0/ljavYbSQAgAJ), if the location of the ring buffer is determinable.

Any thoughts, please?

aralbrec, 2017-02-05:
The linker is not yet able to flow code or data around fixed islands.  This is something being looked at for z80asm2.

You can still do this but there is still a modicum of hand placement in order to minimize wasted space.

**1. Assign memory space not managed by the compiler / linker.**

You can pack items with special alignment requirements by hand into a memory space not managed by the compiler / linker.

Asm file describing assigned items "fixed.asm":

```
SECTION FIXED
org 0xf000

; 256-byte aligned

PUBLIC _circ_buffer_0
_circ_buffer_0:
   defs 256

; 256-byte aligned

PUBLIC _circ_buffer_1
_circ_buffer_1:
   defs 128

PUBLIC _table_0
_table_0:
   defb 1, 2, 3, 4, 5, 6, 7, 8
   defs 30

; no idea where this is but let's align to 256 bytes
; be careful with wasted space
IF (ASMPC & 0xff)
   defs 256 - (ASMPC & 0xff)
ENDIF

PUBLIC _circ_buffer_2
_circ_buffer_2:
   defs 128

; maybe we need to move to a specific address 0xf456
; this may waste a lot of space!

defs 0xf456 - 0xf000 - ASMPC

; let's make sure this area is zeroed at program
; startup - if you need non-zero values you should
; look at compressing the output binary "foo_FIXED.bin"
; and decompressing it at startup instead.

SECTION code_crt_init

ld hl,__FIXED_head
ld bc,__FIXED_size
ld e,0
call asm_memset

; or an ldir variant but this one protects against
; zero size
```

All the public names have leading underscores so they can be seen from C as well as assembler.  In C you might do "extern unsigned char circ_buffer_0[256];" and in asm "EXTERN _circ_buffer_0".

If you add "fixed.asm" to the compile line, you will get a new binary out "foo_FIXED.bin" that corresponds to this section.  You can throw it away if you don't need to initialize that data at startup or if you supply initialization like I did with the memset call above.

ASMPC is special - it is the program counter (corresponding to $ in many other assemblers).  Because this is a linking assembler, its behaviour can be unexpected but it is logical.  When ASMPC is used in a value that is computed at link time, it corresponds to a physical address.  When it is used in a value that is computed at assemble time, it corresponds to the current offset into the section as seen in the current file.

So in "IF (ASMPC & 0xff)" ASMPC is being used at assemble time since IF is an assemble-time decision.  ASMPC starts at 0 at the start of the section and increases as bytes are added in that file - it's therefore an offset from the beginning of the file.  Since the org address was 0x??00, it started 256-byte aligned so as long as the offset ASMPC is also a multiple of 256 bytes, the address there will be 256-byte aligned too.

Similarly in "defs 0xf456 - 0xf000 - ASMPC" ASMPC is an offset because the assembler needs to determine how many bytes to set aside here.  So "0xf456 - 0xf000" tells me the byte offset I need from the beginning of the section and ASMPC tells me the actual byte offset so far.

In hypothetical "defw ASMPC" ASMPC would be the actual address determined at link time.  This is because the assembler can set aside two bytes and say "ASMPC is there" to the linker.

**2. Create a block of buffers at known location in the crt or bss/data sections.**

Similar to (1) but this time you arrange the memory map to hold your items in known sections that become part of the output binary.  So if you have items that should be stored in BSS, make sure they appear first in BSS so that, given the org address of bss (or using a defs offset to guarantee alignment) you can place alignment-sensitive items as in (1).

If the crt is in ram, that's a very good place to put such buffers because you have precise control of placement of everything in the crt.

**3. Use a dynamic memory allocation solution.**

Malloc might be too heavy a solution for what you have planned but the malloc library has functions that allocate on alignment boundaries, see: `void *memalign_unlocked(size_t alignment, size_t size)`.  Of course the more you malloc in this manner, the more the memory pool can become fragmented.  (This function has not seen a lot of testing FYI but if there's a bug we're pretty fast about fixing it).

A lighter solution is the [obstack](https://en.wikipedia.org/wiki/Obstack).  I don't like that explanation too much.  The idea behind an obstack is really simple.  You create a block of memory and set up a fence at the beginning.  As you allocate, the fence moves forward by that many bytes.  Everything before the fence is allocated, any space beyond is available.  You can't free randomly because free just means moves the fence back to a previous address.  So you can free the last item or a bunch of items all at once.  This is why it's called an object stack.  It's lightweight and fast.  obstack also has alignment functions: 

```
; int obstack_align_to(struct obstack *ob, size_t alignment)
;
; Set the obstack fence to the next address aligned to alignment.
; Any following allocations or further growth will begin at this
; address.  Does not close the currently growing object.

; size_t obstack_align_distance(struct obstack *ob, size_t alignment)
;
; Return distance in bytes from the obstack fence to the next
; address aligned to alignment.
```

align_to will move the fence forward to the desired alignment, wasting some amount of space.  align_distance can be used to find out how much space would be wasted ahead of the move.  Interestingly you can also sprintf to the obstack.  There is a notion of building an object on the obstack by incrementally adding stuff to it in the GNU interface and obstack_printf is one of those things.

Those are some ideas - maybe someone else has more to add.

feilipu, 2017-02-06:
Thanks for the complete explanation of options. :+1: 

Noted and bookmarked for use.
I think that option 2 is the most straightforward for me to implement, and it probably will be the path I'll try first.

closed 2017-02-06 by feilipu

aralbrec, 2017-02-08:
A pragma was added "CRT_ENFORCE_SNAMES" that can be used to remove the single zero byte the memory map adds to the beginning of both the BSS and DATA sections.  That way user data can be placed right at the beginning of BSS and DATA without that initial zero byte possibly fouling up alignment set up by the DATA/BSS org.

The commentary in part 2 above has been edited to describe that more fully.

aralbrec, 2017-06-11:
#233 has allowed the introduction of aligned sections for creating data structures in memory at aligned addresses when the DATA and/or BSS sections have an aligned ORG.  This is described in point 2 above.