z88dk/z88dk #59: new c lib (+zx): Bifrost*L and Bifrost*H seems to misbehave at beginning of demos

aralbrec, 2017-02-06:
Two compiled demos for the zx spectrum:

[Bifrost*h demo](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/zx/demo_bifrosth/demo.tap)
[Bifrost*l demo](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/zx/demo_bifrostl/demo.tap)

I'm not sure why some of the tiles seem incorrect.

stefandrissen, 2017-02-06:
I think that your simply attempting to point to tiles that are not there - you are looping through 81 positions and expecting there to be 81 static images:

      for (f = 0; f < 81; ++f) {
            BIFROSTH_tilemap[f] = BIFROSTH_STATIC + f;
        }

aralbrec, 2017-02-06:
I think you are right - there probably aren't 82 tiles defined given the rest of the program is limited to far fewer.  I'll confirm that tomorrow before closing, thanks!

aralbrec, 2017-02-06:
There are only 34 tiles defined for both demos and those are the only ones that correctly display on the initial startup of both demos.  So the engine and program are behaving as expected.

Nothing to see here, move along!

closed 2017-02-06 by aralbrec