z88dk/z88dk #63: z80asm: late computation of org possible to allow placement based on section sizes?

aralbrec, 2017-02-09:
I have an application here that creates a section containing source code and user data.  The goal is to place 
this section at the top of memory but it's unknown ahead of time how big it will be.

This is what I would like to do but I'm not sure if it's possible:

```
SECTION FASE
org $10000 - __FASE_size - __FASE_USER_size

; source code here

SECTION FASE_USER
; user data here appends to section FASE (no org of its own)

```

pauloscustodio, 2017-02-13:
The solution described does not work. The argument to ORG must be a compile-time constant, and the section sizes are determined after linking.

What can be done is a two step process: assemble with split-section to obtain the exact address to start, and then assemble again with -r0xxxxx to set the ORG at that address.

Maybe a future feature for z80asm2 is that sections can be allocated downward from a top maximum address.

aralbrec, 2017-02-17:
> Maybe a future feature for z80asm2 is that sections can be allocated downward from a top maximum address.

I am thinking that the org might not have to be a compile time constant at all... if a list of org expressions are known, the first step in patching could be to try to resolve all org addresses.  Loop through them all repeatedly, see if any org can be resolved to a constant and if no progress can be made, declare an error.

If this is possible one could solve issues like the one raised here in one compile pass instead of two involving manual computation.

I'll try the relinking thing for the current problem and leave instructions to users that a compile has to be two steps.

closed 2017-02-17 by aralbrec