z88dk/z88dk #65: z80asm: has there been a change in treatment of org -1 ?

aralbrec, 2017-02-10:
Has "org -1" been changed in z80asm?

I'm seeing compiles that are no longer generating a separate BSS binary, instead BSS appending to the DATA binary.

In rom compiles, a copy of the data binary has to be stored in the rom so that ram is properly initialized at startup.  The crt zeroes the bss section separately.

By including BSS in DATA a much larger block is stored in rom.

aralbrec, 2017-02-10:
I'll just add a test case:

File "zzz.c"
```
#include <stdio.h>

#pragma output CLIB_OPT_PRINTF = 0

unsigned char buf[100];

void main(void)
{
	sprintf(buf, "Hello World!\n");
}
```

zcc +embedded -vn -clib=new zzz.c -o zzz -create-app -m --list

The "buf" array is placed in BSS as can be seen from the map file but there is no bss binary produced only "zzz_CODE.bin" and "zzz_DATA.bin"

pauloscustodio, 2017-02-12:
I do not recall any changes in the ORG -1 handling.  
There was one change to prune empty sections back in 2016-10-10. Maybe this is the cause?

The test program below correctly generates the BSS section:
```
	section CODE
	ret
	section DATA
	org 0x4000	; split file here
	defw 0
	section BSS
	org -1	; split file here
	defw 1
```
When compiled with ```z80asm -b test.asm```, the following files are generated:

test.bin: ```C9```
test_DATA.bin: ```00 00```
test_BSS.bin: ```01 00```

aralbrec, 2017-02-12:
I found the bug -- it's the intermediate step to object file that causes BSS to disappear.

Same example, this time assembled to object file first:

z80asm xxx.asm
z80asm -b xxx.o

The BSS section is not produced but straight to binary, it is:

z80asm -b xxx.asm

pauloscustodio, 2017-02-12:
Found the root cause. -1 is used in the object file both for "ORG not defined" and "ORG is -1".

commit e9b79de2 (2017-02-12): Fix #65 separate BSS section not generated from object file if BSS org was -1 at assembly time

ORG -1 is used as section split marker, but was not being stored in the
object file, failing to split a section in a separate link step.

Store -2 in the object file in case of section split.

Fix z80nm to give that information.
closed 2017-02-12 by pauloscustodio