z88dk/z88dk #75: sccz80: Allow sccz80 to push arguments r->l

suborb, 2017-02-19:
We can use __smallc to indicate that the arguments are the classic way round which works nicely for libraries.

However, for user code this going to be awkward change.

suborb, 2017-02-20:
At the moment, this works for most cases up to a certain level of nested function calls. The build is failing in savage.c. This following works:

a = exp(log(sqrt(a * a)));

Adding the atan() element fails:

a = atan(exp(log(sqrt(a * a))));

There's a bunch of valgrind errors which I don't think are the cause of the problem.

aralbrec, 2017-02-24:
I have compile errors for vararg functions using sccz80.  Has there been a change in how vararg functions are handled?

Before the change, call to printf:

```
2462  05C4  21 28 00    	ld	hl,i_1+40
2463  05C7  E5          	push	hl
2464  05C8  3E 01       	ld	a,1
2465  05CA  CD 00 00    	call	printf
2466  05CD  C1          	pop	bc
```

After the change call to printf, note number of arguments is no longer in the A register:

```
2461  05C2  21 28 00    	ld	hl,i_1+40
2462  05C5  E5          	push	hl
2463  05C6  CD 00 00    	call	printf
2464  05C9  C1          	pop	bc
```

printf is declared with smallc attribute if that matters:

`extern int __LIB__ printf_unlocked(char *format,...) __smallc;
`

suborb, 2017-02-24:
Oops, the check was the wrong way round so that r2l functions got an a, but l2r didn't.

aralbrec, 2017-02-24:
Code generation is now identical to what it was before the change.

But I have a bug that is probably not related - the sudoku solver cannot solve puzzles under sccz80 compilation but it does under sdcc.  I'm going back in commits to see where this problem started.

commit 439cdcaf (2017-02-19): Fix #75: Capture doubles for function args. Functions still not
handled properly - need a stack of previous states and then I
think it will work..
commit 7429425d (2017-02-19): Fix #75: Sort out the function problem, now having issues when a cast
is involved (Eg atof.c), this may be the main cast problem we have in
sccz80 so worth chasing.
commit f038f673 (2017-02-20): Fix #75: Kludge around the cast issue, and do escaped literals properly.

Library compiles with this version now.

Next: Add a flag to determine default behaviour.
commit 988cce33 (2017-02-20): Fix #75: Introduce flag -set-r2l-by-default to turn r2l handling.
closed 2017-02-24 by aralbrec