z88dk/z88dk #90: sccz80: all primitive comparison functions bugged

aralbrec, 2017-02-26:
Following discovery of bug #82 where l_eq was found to be required to set the result of hl=0/1 as well as the carry flag, other comparison functions were also checked:

```
unsigned int a, b, c;

void main(void)
{
	c = (a==b)*2;
	c = (a<=b)*2;
	c = (a<b)*2;
	c = (a>=b)*2;
	c = (a>b)*2;
}
```

Asm translation:

```
._main
	ld	de,(_a)
	ld	hl,(_b)
	call	l_eq
	add	hl,hl
	ld	(_c),hl
	ld	de,(_a)
	ld	hl,(_b)
	call	l_ule
	add	hl,hl
	ld	(_c),hl
	ld	de,(_a)
	ld	hl,(_b)
	call	l_ult
	add	hl,hl
	ld	(_c),hl
	ld	de,(_a)
	ld	hl,(_b)
	call	l_uge
	add	hl,hl
	ld	(_c),hl
	ld	de,(_a)
	ld	hl,(_b)
	call	l_ugt
	add	hl,hl
	ld	(_c),hl
	ret
```

In each case, hl is again expected to hold 1/0 if the condition is true/false.  The current implementation of integer comparison do not do that but it appears long comparison might be ok.

For the simpler case where the compiler only uses the carry flag of the comparison result:

```
unsigned int a, b, c;

void main(void)
{
	c = (a==b);
	c = (a<=b);
	c = (a<b);
	c = (a>=b);
	c = (a>b);
}
```

Translated asm:

```
._main
	ld	de,(_a)
	ld	hl,(_b)
	call	l_eq
	ld	hl,0	;const
	rl	l
	ld	(_c),hl
	ld	de,(_a)
	ld	hl,(_b)
	call	l_ule
	ld	hl,0	;const
	rl	l
	ld	(_c),hl
	ld	de,(_a)
	ld	hl,(_b)
	call	l_ult
	ld	hl,0	;const
	rl	l
	ld	(_c),hl
	ld	de,(_a)
	ld	hl,(_b)
	call	l_uge
	ld	hl,0	;const
	rl	l
	ld	(_c),hl
	ld	de,(_a)
	ld	hl,(_b)
	call	l_ugt
	ld	hl,0	;const
	rl	l
	ld	(_c),hl
	ret
```

The compiler is inserting code to generate hl=1/0 itself.

The primitives need to be changed to set hl=1/0 on comparison result in addition to setting the carry flag as it does now.  After this change the compiler will no longer have to generate code like the above either as it will be done inside the primitive routine.

aralbrec, 2017-02-26:
The changes have been made.  The test program I used is below and was run testing int, unsigned int, long, unsigned long.

```
// zcc +zx -v -O3 zzz.c -o zzz -lndos -create-app --list
// zcc +zx -v -O3 -clib=new zzz.c -o zzz -create-app --list

#include <stdio.h>

#define TEST   unsigned long
#define TESTPF "%lu"

TEST a, b;

void comp(TEST a, TEST b)
{
	printf("\n\nCompare a=" TESTPF ", b=" TESTPF "\n", a, b);
	printf("a==b : %d\n", (a==b)*2);
	printf("a<=b : %d\n", (a<=b)*2);
	printf("a< b : %d\n", (a<b)*2);
	printf("a>=b : %d\n", (a>=b)*2);
	printf("a> b : %d\n", (a>b)*2);
	printf("a!=b : %d\n", (a!=b)*2);
	printf("!a   : %d\n", (!a)*2);
}

void main(void)
{
	while (1)
	{
		printf("\n\nEnter a: ");
		scanf(TESTPF, &a);
		printf("Enter b: ");
		scanf(TESTPF, &b);
		comp(a,b);
	}
}
```

closed 2017-02-26 by aralbrec

suborb, 2017-02-27:
The cause of this is actually in plng2a(). The carry expression is correctly expanded out to an INT using the rl l code, but because it's a constant multiplication the stage is cleared and we call quikmult().

I'm not sure what's the best course of action to do here: if we set hl in the primitives then we're incurring extra T states to cope with this possibility. On the other hand, keeping with the rl l route we use extra space, but only when needed.

commit 8c03154a (2017-02-26): Fix #90: sccz80 comparison primitives now return hl=1 or 0 depending on true/false state of result
commit 8194f4c5 (2017-02-27): Fix #90: Don't throw away the result of a carry conversion