z88dk/z88dk #92: classic: compressed rom compile bugged

aralbrec, 2017-03-01:
Another issue coming out of:
https://www.z88dk.org/forum/viewtopic.php?pid=14896#p14896

A compressed rom compile is behaving differently from an uncompressed one.

suborb, 2017-03-06:
I've just tried this out the +z88 application target and I'm not seeing an issue, compressed ROM behaves as expected. So I suspect it's port specific.

aralbrec, 2017-03-10:
I can't see anything wrong with the initialization on that msx program either -- it might be necessary to see if there is something wrong with the program itself.

aralbrec, 2017-04-09:
More update on this bug:
https://www.z88dk.org/forum/viewtopic.php?pid=14979#p14979

There is something wrong with the uncompressed rom's ram initialization (it works for the compressed model but not the uncompressed model) and there may be a bug in sccz80.

aralbrec, 2017-04-24:
"Port fixing" means setting the hardware memory configuration.  This has to come first in the msx crt for cartridges and we had it after the bss/data initialization.

commit 15c8ebbf (2017-04-24): Fix #92: bss and data section intiialization must happen after the hardware memory configuration
closed 2017-04-24 by aralbrec