z88dk/z88dk #95: classic: problem with enabling %ld and %e for printf

aralbrec, 2017-03-03:
I ran into this problem while compiling Whetstone with the classic library.

The output is not displayed properly as if %ld and %e are not enabled for printf (the output shows "ld" and "e" in the printed text).

[Overview of whetstone](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/whetstone/readme.txt)
[Compile instructions for z88dk](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/whetstone/z88dk/readme.txt)

The [whetstone source code](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/whetstone/z88dk/whetstone.c) can be compiled from:
z88dk/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/whetstone/z88dk

A specific compile line targetting the zx spectrum using sccz80/genmath:
`zcc +zx -vn -O2 -DSTATIC -DTIMER -DPRINTOUT whetstone.c -o whetstone -lm -lndos -m -create-app`
You will see the incorrectly printed results as output when run.

A correct compile targetting the zx spectrum using zsdcc/math48 for reference in this [tap file](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/whetstone/z88dk/whetstone.tap).

In the whetstone source code I am enabling %ld and %e (among other converters) with:

```
#ifdef __Z88DK
   #include <intrinsic.h>
   #ifdef PRINTOUT
      #pragma output CLIB_OPT_PRINTF = 0x05001001
   #endif
#endif
```

__Z88DK is defined in zx.cfg

suborb, 2017-03-03:
I can't try this out at this point in time.

However, sccz80 thinks that printf flags should be 0x41001000 - i.e. flag handling is required for printf. Your override might be causing issues with printf when parsing the flags in the format string.

aralbrec, 2017-03-03:
I tried "CLIB_OPT_PRINTF = 0x41001000" but the program crashes and goes blank.

Doublechecking against "crt0_runtime_selection.asm" the bit assignments look the same as the new c lib so I'm not sure why there should be a different bit string?

The converters required are: %d, %ld, %e, %f

I tried "CLIB_OPT_PRINTF = 0x45001001" just in case the initial 0x40000000 is what you were on about.  This also crashes with blank screen.

A hail mary with "CLIB_OPT_PRINTF = 0x7fffffff" causes a crash as well.

It could be a bug in the %ld converter as it crashes immediately after printing the first character and before the first %ld conversion.

suborb, 2017-03-04:
It looks like it might be a problem with the width specifier supplied to %e or any floating point operator:

This:

    int main()
    {
        long d = 1000000L;
        double e = 1.23;
        printf("Hello world! %7ld %12.4e\n",d,e);
    }

When compiled with CPM generates pages of output before the floating point number.

The difference in the printf pragma is because I made flag handling an option within printf. %d is only encountered when -DTIMEFUNC is supplied

suborb, 2017-03-04:
I've tracked this down to __printf_print_aligned - there's a call to l_ge which assumed that hl is left undisturbed. The fix for #90 made the comparison routines set hl which broke that assumption.

commit 0cdc14df (2017-03-04): Fix #95: Preserve hl over the call to l_ge
closed 2017-03-04 by suborb