z88dk/z88dk #97: z88dk/z80asm support for 8080 syntax - enhancement

feilipu, 2017-03-06:
Somewhat related to issue #64, I'm keen to integrate an 8080 syntax floating point library into z88dk.

The LLL floating point library is available in Intel 8080 syntax and comes to us [directly from 1975](https://github.com/feilipu/LLL-Floating-Point).

Wondering what is the best way to incorporate 8080 syntax code into the z88dk, without the laborious and error prone method of transcribing into z80 syntax?

aralbrec, 2017-03-06:
There are some automatic translators around that will do 8080 -> z80 mnemonics.

There's even one included in z88dk implemented as an [awk script](https://github.com/z88dk/z88dk/tree/master/support/8080).

I've never used it before so I'm not sure how good a job it does.

feilipu, 2017-03-06:
That looks like a good starting point.
Good enough to get the bulk of the work done, anyway.
Thanks.

closed 2017-03-06 by feilipu

feilipu, 2017-03-06:
For completeness, the 8080 -> z80 transcription process by the awk script was [almost flawless](https://github.com/feilipu/LLL-Floating-Point/commit/b30d59f84afe3667187d0e9c5634b93ae3c00ed0).

Two minor errors remained.
`LD A, M` instead of `LD A,(HL)`
`SBC (HL)` instead of `SBC A,(HL)`

With these two corrections identical HEX was generated.
Success.