z88dk/z88dk #100: zsdcc: segmentation fault for specific use of __smallc function attribute

aralbrec, 2017-03-10:
Also reported as [sdcc bug 2590](https://sourceforge.net/p/sdcc/bugs/2590/).

test.c
```
#define MODE2_ATTR  8192

extern const unsigned char letras_tiles01[208];

extern void msx_vfill(unsigned int addr, unsigned int value, unsigned int count) __smallc;
extern void msx_vwrite_direct(void* source, unsigned int dest, unsigned int count) __smallc;

void poner_texto_tiles(unsigned char *texto, unsigned int size, unsigned int posicion, unsigned char color)
{
      unsigned char i = 0;

      while (size)
      {

            if (texto[i] == ' ')
            {
                  msx_vfill((posicion << 3), 0x00, 8);
                  msx_vfill(MODE2_ATTR + (posicion << 3), color, 8);
            }
            else
            {
                  msx_vwrite_direct(letras_tiles01 + ((*(texto + i) - 65) << 3), posicion << 3, 8);
                  msx_vfill(MODE2_ATTR + (posicion << 3), color, 8);
            }

            i++;
            posicion++;
            size--;
        }
}
```

Translation to assembly:
zcc +zx -v -a -clib=sdcc_iy test.c

Result is a segmentation fault.

I traced this to the msx_vfill() calls in the true part of "if (texto[i] == ' ')". If I comment out this two calls, compilation is successful including the compilation of an identical call to msx_vfill() in the false part of the "if".

I also found that eliminating the smallc attribute on the msx_vfill() prototype leads to successful compilation. This leads to the suspicion that smallc is somehow bugged.

aralbrec, 2017-03-13:
Rebuilt zsdcc windows binary is available from:
https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_z88dk_patch.zip

@suborb 
The osx zsdcc binary will have to be rebuilt and the nightly build packages updated.

The windows build is at sdcc 9852

closed 2017-03-13 by aralbrec

suborb, 2017-03-13:
Updated, next nightly should have new binaries.