z88dk/z88dk #98: sccz80 bug: incorrect stack offset for a long and function epilogue messed up

aralbrec, 2017-03-10:
This one comes while trying to compile the binary-trees benchmark.

Test program:

```
#define NULL  ((void *)0)

typedef struct tn {
    struct tn*    left;
    struct tn*    right;
    long          item;
} treeNode;

extern treeNode* NewTreeNode(treeNode* left, treeNode* right, long item);

treeNode* BottomUpTree(long item, unsigned depth)
{
    if (depth > 0)
        return NewTreeNode
        (
            BottomUpTree(2 * item - 1, depth - 1),
            BottomUpTree(2 * item, depth - 1),
            item
        );
    else
        return NewTreeNode(NULL, NULL, item);
} /* BottomUpTree() */
```

Translation to assembly:

zcc +zx -vn -a -O2 -clib=new zzz.c --list --c-code-in-asm

Looking at just the ELSE part of the IF:

```
;else
.i_3
;return NewTreeNode(((void *)0), ((void *)0), item);
	LINE	22
;((void *)0);
	ld	hl,0	;const
	push	hl
;((void *)0);
	ld	hl,0	;const
	push	hl
;item;
;;;**************************
	ld	hl,65528	;const
	add	hl,sp
	call	l_glong
	push	de
	push	hl
	call	_NewTreeNode
	pop	bc
	pop	bc
	pop	bc
	pop	bc
	exx
	ld	hl,65520	;const
	add	hl,sp
	ld	sp,hl
	exx
	ret
```

At the point indicated by ************ the stack frame looks like:

(higher stack address)
item (4 bytes)
depth (2 bytes)
return address (2 bytes)
0 (2 bytes)
0 (2 bytes)  <--- SP

The next thing that is supposed to be fetched and stored on the stack for the function call is "item".  This is now at SP+8 but the offset used is "65528" which is -8 not +8.

Another error occurs after the call to _NewTreeNode.  The following four "POP BC" clear the stack up to the return address.  Instead of returning, this code:

```
	exx
	ld	hl,65520	;const
	add	hl,sp
	ld	sp,hl
	exx
	ret
```

is run to "clear the stack" which manages to corrupt it by erroneously adjusting SP downward by 16 bytes.

suborb, 2017-03-10:
I think this is caused by the r2l changes. It looks like although we clear the assembler that's generated when we collect the arguments to a function call, the stack offset isn't being reset. There's an easy fix for it (save Zsp before the first while() loop in callfunc and restore afterwards) which I'll push through this evening, but I also want to investigate why it's happening.

aralbrec, 2017-03-10:
There has been a change but the generated code is still not right.

For the same bit of code:

```
203   0119              .i_6
204   0119  21 00 00    	ld	hl,0	;const
205   011C  E5          	push	hl
206   011D  E5          	push	hl
;;;;;;******************************************
207   011E  21 04 00    	ld	hl,4	;const
208   0121  39          	add	hl,sp
209   0122  CD 00 00    	call	l_glong
210   0125  D5          	push	de
211   0126  E5          	push	hl
212   0127  CD 00 00    	call	_NewTreeNode
213   012A  C1          	pop	bc
214   012B  C1          	pop	bc
215   012C  C1          	pop	bc
216   012D  C1          	pop	bc
217   012E  C5          	push	bc
218   012F  C5          	push	bc
219   0130  C9          	ret
```

At ************ the stack frame looks like:

(higher stack address)
item (4 bytes)
depth (2 bytes)
return address (2 bytes)
0 (2 bytes)
0 (2 bytes) <--- SP

"item" is being retrieved next but it's at offset +8 not +4.

The four "pop bc" after the call to _NewTreeNode completely clear the stack but the compiler thinks it needs to push two more items to level the stack.

There is a sign error somewhere where the number of bytes on the stack is being accumulated.  A plus instead of a minus or the other way around.

suborb, 2017-03-11:
Sorry about that, the usecase to reproduce it is actually really simple:

    long func(long l)
    {
        return 2 * l;
    }

The code being generated is obviously total garbage (as a quick workaround to get you going on the benchmarks, use 2L or put the constant second.

suborb, 2017-03-11:
Need to add some more testcases for correct promotion.

commit b82fb5fa (2017-03-10): Fix #98: Restore the stack after we've collected the arguments.
commit 54250a23 (2017-03-11): Fix #98: Some tests to verify int.

Doubles will be rolled into 103.
closed 2017-03-12 by suborb