z88dk/z88dk #103: sccz80: unable to use integer constraints as control on float variable in for loop

aralbrec, 2017-03-11:
This one comes from the mandelbrot benchmark.

Test code:

```
double y;

void main(void)
{
	for (y=0; y < 10  ; y+=1.0) ;
	for (y=0; y < 10.0; y+=1)   ;
	for (y=0; y < 10.0; y++)    ;
}
```

The assignment "y=0" does not seem to be a problem.
The for-loop test causes an error if it is an integer.  "y<10" fails but "y<10.0" is ok.
The increment does not work unless it is float.  "y+=1.0" is ok, "y+=1" fails and "y++" fails.

suborb, 2017-03-11:
At least this one errors rather than generate bad code. I'll take a look.

suborb, 2017-03-11:
Actually there were code generation errors, from inspection and running, the code in that branch should now be correct.

The post-(in|de)crement operator isn't optimised so will end up doing two fp additions for each operation.

closed 2017-03-12 by suborb