z88dk/z88dk #106: ticks: Pass through command line arguments to the z80 program

suborb, 2017-03-12:
Collate as a space separated string and place at a known point in memory.

Probably makes sense to stick them at 65280 or so and then have the +test target move its stack to below them.

All arguments after a "--" option should be passed through to the z80 machine.

Doing this should allow use of +test as a general purpose development target.

aralbrec, 2017-03-12:
In the most recent changes to ticks, alarm() is being used to set a timeout value.  The problem with this is there is no alarm() in the win32 api and we'd have to use some win32 code to set up a timer.  This is not impossible but unless you need some asynchronous interrupt, is it adequate to use ticks' built-in "-counter X" ?  This will stop ticks after X z80 cycles have run where X is a 64-bit unsigned long long (so in excess of many years real time execution on a 4MHz z80).

suborb, 2017-03-12:
Ok, I've changed it to use an approximate count of 400000000LL cycles per second. 

It's only there as a failsafe for when something goes badly wrong in the test, e.g. infinite loop, or di/halt which will now not be worked around.

commit df5dd67a (2017-03-13): Fix #106: Pass any arguments supplied after -- to ticks through to the target.

Upto 255 characters allowed to passed through, length of arguments is put
at 65280, the arguments are placed from 65281.
closed 2017-03-13 by suborb