z88dk/z88dk #108: math48 (sccz80): Incorrect results for sqrt()

suborb, 2017-03-12:
Possibly related to the dswap() Issue #104 fix?

    void run_sqrt(double x, double e)
    {
        double r = sqrt(x);
        printf("Sqrt(%f) should be %f but was %f\n",x,e,r);
    }

    int main() 
    {
        double x = 4.0;
        double e = 2.0;
        double r = sqrt(x);
        printf("Sqrt(%f) should be %f but was %f\n",x,e,r);
        run_sqrt(4.0, 2.0);
        run_sqrt(9.0, 3.0);
        run_sqrt(1.0, 1.0);
    }

Yields the results:

    Sqrt(4.000000) should be 2.000000 but was 0.000000 
    Sqrt(4.000000) should be 2.000000 but was 0.707107
    Sqrt(9.000000) should be 3.000000 but was 191.977321
    Sqrt(1.000000) should be 1.000000 but was 213.965035

Building with sdcc gives the correct results. Genmath works as expected too.

aralbrec, 2017-03-12:
It's working as sccz80/newclib.  I'll try a compile under sccz80/classic after building the libs.

aralbrec, 2017-03-12:
Yes working under the new c lib but not the classic.

```
#include <stdio.h>
#include <math.h>

#pragma output CLIB_OPT_PRINTF = 0xffffffff

void run_sqrt(double x, double e)
{
    double r = sqrt(x);
    printf("Sqrt(%f) should be %f but was %f\n",x,e,r);
}

int main() 
{
    double x = 4.0;
    double e = 2.0;
    double r = sqrt(x);
    printf("Sqrt(%f) should be %f but was %f\n",x,e,r);
    run_sqrt(4.0, 2.0);
    run_sqrt(9.0, 3.0);
    run_sqrt(1.0, 1.0);
}
```

sccz80/new
zcc +zx -vn -startup=4 -clib=new zzz.c -o zzz -create-app -lm

sccz80/classic
zcc +zx -vn zzz.c -o zzz -create-app -lmath48 -lndos

aralbrec, 2017-03-12:
The problem is the math.h headers are different for the classic and new c libraries.  The math48 math functions that take a single double parameter are declared FASTCALL whereas in the classic lib they are not.

This program will work under sccz80/classic:

```
#include <stdio.h>

extern double_t __LIB__ __FASTCALL__ sqrt(double_t x);

#pragma output CLIB_OPT_PRINTF = 0xffffffff

void run_sqrt(double x, double e)
{
    double r = sqrt(x);
    printf("Sqrt(%f) should be %f but was %f\n",x,e,r);
}

int main() 
{
    double x = 4.0;
    double e = 2.0;
    double r = sqrt(x);
    printf("Sqrt(%f) should be %f but was %f\n",x,e,r);
    run_sqrt(4.0, 2.0);
    run_sqrt(9.0, 3.0);
    run_sqrt(1.0, 1.0);
}
```

where instead of including math.h I've manually inserted the prototype for sqrt() making it fastcall.

The [sccz80 math.h](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/sccz80/math.h) header from the new c library where single parameter math functions are made fastcall.

The [sccz80 math.h](https://github.com/z88dk/z88dk/blob/master/include/math.h) header from the classic c library.

I'm not sure how difficult it is yet to get fastcall for math48 in the classic header.

suborb, 2017-03-12:
That's subtle, I didn't even realise that fastcall worked for doubles! 

 sccz80 will call dload() in the fastcall case and dldpsh() (or really dload/dpush) in the non fastcall. So if we have dldpsh do an exx then I think we get the same behaviour as per ints/longs - the primary register is setup with the value.

aralbrec, 2017-03-12:
There are more inconsistencies for the multiple parameter cases.  For example, atan2() and pow() are implemented in asm in math48 but in c in genmath.  In math48's case these are made callee and in genmath's case this can't be done.  I think I did leave the standard linkage for these functions in math48 so we might be able to gloss over these with the change you suggest but we lose a little bit on preformance and code size.

aralbrec, 2017-03-12:
`sccz80 will call dload() in the fastcall case and dldpsh() (or really dload/dpush) in the non fastcall. So if we have dldpsh do an exx then I think we get the same behaviour as per ints/longs - the primary register is setup with the value.`

The dldpsh implementation for math48 is wrong... it's only pushing the float onto the stack instead of also loading it into the primary accumulator.  So I'll fix that up and it should work.

aralbrec, 2017-03-12:
The fix is in the dev/aralbrec branch but it is causing a crash for printf

commit cc4da0ed (2017-03-12): Fix #108: math48 dldpsh must also load the value into the primary float accumulator
commit 629f659c (2017-03-12): Fix #108: printf_handle_f no longer uses dldpsh; there is no way to use that function in that manner that would satisfy all combinations of compiler, math lib and restricted targets
closed 2017-03-12 by aralbrec