z88dk/z88dk #109: sccz80: Callee cleanup of functions destroys return value.

suborb, 2017-03-12:
int run_sqrt(double x, double e) __z88dk_callee
    {
        double r = sqrt(x);
        printf("Sqrt(%f) should be %f but was %f\n",x,e,r);
        return 1;
   }

Generated function prologue:

        ld      hl,1    ;const
        pop     bc
        pop     bc
        pop     bc
        pop     de
        ld      hl,18   ;const
        add     hl,sp
        ld      sp,hl
        push    de

aralbrec, 2017-03-12:
I don't know if this is related but some time ago I put in a fix to avoid altering the exx set when the return type is double:

https://github.com/z88dk/z88dk/commit/09fbdfcef5aaf955dd71e86d0b186bd8a561ed41#diff-29340938e68a2ca908ff6c1de35c971f

suborb, 2017-03-12:
That looks like the cleanup after a function call.

This is for cleaning up the stack inside a function. Doubles were okay, but ints/longs were being trashed.

commit 341e7bff (2017-03-13): Fix #109: Add a rule to cleanup a double exx
commit 5d156447 (2017-03-13): Fix #109: Cleanup of callee functions was trashing the return values. Sort
it out. Clean up the noaltreg paths.

Preserve the exx set if a double is being returned (for math48 case)
closed 2017-03-13 by suborb