z88dk/z88dk #111: z88dk: benchmarks

aralbrec, 2017-03-14:
I've been collecting a set of [compiler benchmarks](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks) in the repository and will be running them against several z80 c compilers for comparison purposes.  This helps us out by highlighting where things can be improved and inspection of the code generated might identify improvements that can be made for common code fragments.  It also gets the word out about just how good z88dk is currently in comparison.

### LIST OF COMPILERS

**Hitech C v3.09 CPM**
Hitech's last CPM C compiler.  It runs on the z80 itself and is surprisingly good.  This compiler represents the best z80 native code generator.  Hitech made this available for free many years ago.

**Hitech C v7.50 MSDOS**
Hitech's second last z80 compiler running on MSDOS (v7.80 was their last).  This compiler disappeared along with Hitech's non-PIC compilers after Hitech was bought by Microchip.

**IAR Z80 v4.06A**
IAR's last z80 compiler running under windows.  Although it's not currently listed on their webpage, they are willing to sell it to anyone who has the cash.

**SDCC**
Open source optimizing compiler for several CPUs including the z80.

**Z88DK**
(of course).  All combinations of the two c compilers (sccz80 & zsdcc), the two c libraries (new & classic) and the several math libraries (primarily genmath and math48).

What I am missing and hoping someone will step forward with in this lifetime:

**Zilog ZDS I v3.68**
Zilog's last z80 compiler.  Completely unavailable - Zilog will not sell it but they are willing to send binaries to anyone who bought a license way back when.  What they offer for free download is the IDE with assembler and linker but no compilers.  ZDS II for free download will produce ez80 code but unfortunately only for the ADL mode and not the z80 mode (24-bit registers, large memory space, special instructions).

**Softools Control Cross-C**
Still available for sale.  Special in that it can generate automatically bankswitched code for the z180.

### LIST OF BENCHMARKS

The standard synthetic ones:

Dhrystone 2.1
Whetstone 1.2
Coremark (license prevents source code from being publicly available)

From the benchmarks game website:

binary-trees
fannkuch
fasta
mandelbrot
n-body
spectral-norm

Common ones:

pi
sieve

I'm also looking to add a "gamer's benchmark" containing common operations for games, which is the most common use for z80 c compilers these days.

The last benchmark will be program size and I will likely try to compile programs for the lowest common denominator CP/M target from the z88dk examples directories.

Some results are already in the benchmarks directory above (click on the benchmark and read the readme) but I will be running them again once the source code for all the benchmarks are worked out.

Does anyone have any other benchmarks to add?

aralbrec, 2017-03-22:
A couple more benchmarks have been suggested:

1. sorting - examine the library supplied qsort
2. sprintf/sscanf - test speed of stdio front end.

Anyone volunteer for the excruciatingly annoying sprintf/sscanf testing? 📦

aralbrec, 2017-04-10:
Another benchmark added:  aes256 which tests logical operators.

suborb, 2017-04-25:
I've just re-run a couple of the benchmarks with sccz80/classic and the ticks count has decreased for them.

This may be a result of either code generation changes or better optimisation rules. 

All benchmarks probably ought to be run with -sccz80-inline-ints which causes the l_gint and l_pint calls to be inlined.

As an aside, it looks like -O3 makes things worse, so stick with -O2

spth, 2017-10-26:
Can any results be found anywhere? I mostly wonder how SDCC performs vs. the others in scores and code size.

Philipp

suborb, 2017-10-26:
There's some results in z88dk/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/*/readme.txt from AA's first run.

Things will have changed since then with both sdcc and sccz80

spth, 2018-05-18:
You might want to add stdcbench (www.stdcbench.org), as it has been designed to work on the kind of systems that support Dhrystone/Whetstone/Coremark, and is not as biased towards some narrow functionality as those.

Philipp

feilipu, 2022-06-15:
Follow this in #1621, where perhaps STDC Bench will be added.

closed 2022-06-15 by feilipu