z88dk/z88dk #113: libraries: malloc / free errors

aralbrec, 2017-03-15:
While running the binary-trees benchmark I found that the classic lib's malloc implementation was much faster than the new c lib's.  That's strange because the new c lib's malloc implementation should be pretty fast.

So I wrote a test program [malloc_test.c](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/malloc_test.c) to get to the bottom of things.  This test program performs 20000 malloc/free operations with varying size allocations so that the heap should be sufficiently fragmented to test edge cases.

What I found is that the classic lib's implementation has many errors, to the point that eventually memory corrupts and the test program crashes.  The new c lib has 2 errors in an sdcc compile and 1 error in an sccz80 compile.  Both compiles get the same random number sequence generated so the discrepancy must be down to compiler error in one or the other.  I'm hoping that's all it is rather than a tiny bug in the malloc implementation itself.

To compile for the classic lib, comment out the include of stropts, the ioctl() and the pragma (classic pragma selection for %u isn't working btw).

zcc +zx -vn -O3 malloc_test.c -o malloc_test -lndos -lmalloc -pragma-define:USING_amalloc -create-app -DPRINTF
zcc +zx -vn -compiler=sdcc malloc_test.c -o malloc_test -lndos -lmalloc -pragma-define:USING_amalloc -create-app -DPRINTF

This is rather surprising as these malloc implementations have been in use a long time in released programs without seeming to have any bugs.

aralbrec, 2017-03-15:
There appear to be serious problems with malloc in the z80 compilers I have tested.

For the binary-trees benchmark,

zsdcc/new takes 27 minutes
sdcc takes 17 seconds
hitech c v750 takes 60 seconds

A modern computer is probably 0.007 seconds.

I've had enough for tonight but it looks like some fun ahead :(
First thing I'll do is find out if any of the compilers pass malloc_test.c

suborb, 2017-03-15:
There is something odd going on there. I can get the test through to the end (with errors) by changing:

      vic = (unsigned char)((long)rand() * N / RAND_MAX);

with 
  
      vic = rand() % N;

With the small range 0-32767 then 32767 will be picked at some point by rand().

I'll look at the errors now.

suborb, 2017-03-15:
BTW, binary-tree works on classic/sccz80, gets a counter value of about 320,000,000 with sdcc < 100,000,000.

aralbrec, 2017-03-16:
> There is something odd going on there. I can get the test through to the end (with errors) by changing:
> `vic = (unsigned char)((long)rand() * N / RAND_MAX);`

Yep this is wrong and cause N to be returned which is even likely for a run of 20000 allocations with RAND_MAX = 32767.  It should be (RAND_MAX+1) that is in the divisor but even that is a minefield because a signed (RAND_MAX+1) can equal -1.

I changed it to:
`vic = (unsigned char)((unsigned long)rand() * N / ((unsigned int)RAND_MAX + 1));`

As you pointed out memrchr() was bugged in the classic lib and you've fixed that.

Now everything passes, both libraries and both compilers so there is no bug in the z88dk malloc implementations.

I'm checking now why sdcc's malloc implementation is faster when it's expected to be slower.

aralbrec, 2017-03-16:
The sdcc compile of binary-trees is bugged - that's why it's so fast.  sdcc's malloc implementation checks out ok.

The classic lib compile of binary-trees is coming out 20 times faster than the new c lib.  I'm not sure how that is possible.  The basic malloc loop in classic is 122 cycles and its free is O(n).  The new c lib has a basic malloc loop of 209 cycles with an O(1) free.  sdcc's malloc loop is 467 cycles by comparison.  All malloc implementations are first fit.

aralbrec, 2017-03-16:
malloc_test.c times at about the same speed for classic and new c libs so the problem appears not to be in the malloc lib at all.

suborb, 2017-03-16:
I've done some optimisation work in sccz80 for long multiplication. Handling the *2 case sped things up considerably for this benchmark (to ~175) and is a general win in terms of space/speed.

aralbrec, 2017-03-16:
I think it's an algorithmic problem.  There are no bugs.

Although both the new malloc and the classic malloc are first-fit, the classic malloc only holds a list of free blocks whereas the new malloc has a list of all blocks, allocated or not.  This was done to make realloc very fast to support the vector<char> type in the library that is used for things like getdelim() and memory streams.  It also means to find a free block, the new malloc is searching through allocated blocks as well as free.  In the binary-trees benchmark, there are a lot of small blocks (maybe thousands) so the new malloc is searching through a lot of allocated blocks to find space.  This penalty is not showing up in the malloc_test program because allocation sizes are widely variable so there isn't a huge number of blocks in the heap at any one time.

The 20x performance difference on this benchmark is just too much to accept even though the advantage of fast realloc is important to the library.  I'll have to see if a new implementation can improve on things.

closed 2026-10-14 by feilipu