z88dk/z88dk #118: sccz80: unable to place doubles in array initializer list

aralbrec, 2017-03-19:
This one is from the benchmark program fasta.c.

```
struct aminoacids {
    char c;
    double p;
};

struct aminoacids iub2[] = {
    { 'a', 0 },
    { 'c', 1 },
    { 'g', 2 },
    { 't', 3 },
    { 'B', 4 },
    { 'D', 5 },
    { 'H', 6 },
    { 'K', 7 },
    { 'M', 8 },
    { 'N', 9 },
    { 'R', 0 },
    { 'S', 1 },
    { 'V', 2 },
    { 'W', 3 },
    { 'Y', 4 }
};

/*
struct aminoacids iub[] = {
    { 'a', 0.27 },
    { 'c', 0.12 },
    { 'g', 0.12 },
    { 't', 0.27 },
    { 'B', 0.02 },
    { 'D', 0.02 },
    { 'H', 0.02 },
    { 'K', 0.02 },
    { 'M', 0.02 },
    { 'N', 0.02 },
    { 'R', 0.02 },
    { 'S', 0.02 },
    { 'V', 0.02 },
    { 'W', 0.02 },
    { 'Y', 0.02 }
};
*/
```

The commented out portion causes errors so I tried the first version above with integer parameters instead.  This passes a compile but the integer constants are not converted to double, instead the array is initialized with integers of size 2 bytes instead of doubles of size 6.

```
._iub2
	defb	97
	defw	0
	defb	99
	defw	1
	defb	103
	defw	2
	defb	116
	defw	3
	defb	66
	defw	4
	defb	68
	defw	5
	defb	72
	defw	6
	defb	75
	defw	7
	defb	77
	defw	8
	defb	78
	defw	9
	defb	82
	defw	0
	defb	83
	defw	1
	defb	86
	defw	2
	defb	87
	defw	3
	defb	89
	defw	4
```

suborb, 2017-03-19:
The first case is handled in the commit above (as long as -doublestr isn't on). The second case (actual floating point numbers) is much deeper. Double expressions aren't folded during the compile (e.g. 1.0 + 0.2 invokes the target dadd function). 

It may take me a while to figure out how to sort this one out unfortunately.

aralbrec, 2017-03-20:
The same problem exists for initializing static doubles inside a function.
An error occurs for variable "limits" declared in main() for the benchmark program mandelbrot.

aralbrec, 2017-03-20:
A possible solution might be to put initializer code into "section code_crt_init" when doublestr is enabled.  That can lead to a load of extra code so doing that when doublestr is not enabled would be bad as just evaluating the constant in the compiler and storing 6 bytes would be better.

suborb, 2017-03-20:
> A possible solution might be to put initializer code into "section code_crt_init" when doublestr is enabled

Good idea, I'm not sure whether anybody uses -doublestr, but it's good for quickly bringing up new math libraries without changing sccz80.

> The same problem exists for initializing static doubles inside a function.

Yup, it's the same code path, I've got them initialising now, as long as they're whole numbers, so just a few more tweaks and you'll be able to run those benchmarks.

Expression folding for doubles now works, which speeds things up and may well reduce the size of the final binary.

suborb, 2017-03-20:
Okay, if you want to experiment with this branch it looks like it's working for mandelbrot and fasta now (there's a trailing comma in one of the initialisations that it's not happy about).

fasta is definitely generating the right output, not sure how to prove mandelbrot though.

Still to do:

* -doublestr handling
* I'll fix up the trailing comma initialisation false error

aralbrec, 2017-03-21:
I'll go back and rerun the affected benchmarks once I've made a single pass through them all.  These things take some time to compile for the various compilers because many of them are not easy to use so I'm trying to get through them all as fast as I can.

A new variation is the recently finished n-body.  This one has constant double expressions in a static array.

suborb, 2017-03-24:
Fixed in PR #126

closed 2017-03-24 by suborb