z88dk/z88dk #120: z88dk: implement a fast ieee 32-bit float type

aralbrec, 2017-03-20:
I may do this at some point but this could be a project taken on by anyone interested.
There were some surprising results in the benchmarks related to this.

Hitech C v309 for CP/M (a native z80 compiler) has by far the fastest 32-bit float implementation.  Its performance best manifested in the [mandelbrot benchmark](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/mandelbrot) with an execution time almost half that of the next 32-bit implementation.  z88dk's float library is 48-bit so you can't quite compare apples to apples but even taking into consideration the 24-bit mantissa vs 40-bit, the hitech implementation is still likely a faster implementation.  In any case, it shows that a 32-bit library can be significantly faster.

The problem with the Hitech implementation is that it is not usable for anything except add, subtract, multiply and divide.  The transcendental functions have large errors associated with them.

How is it so fast?  I think the multiply and divide operations are being done with integer math.  Both of z88dk's 48-bit float implementations are preferring to keep the mantissa in a sequence of registers for which multiply and divide are done with sub-optimal shift & add/sub.  It's sub-optimal because holding values in certain registers allows integer math to be much faster for the same operations.  The other compilers' 32-bit math implementations barely outperform z88dk's 48-bit which tells me they are doing something similar.

z88dk has some very fast integer math, especially in the new c library which has a fast math option.  This integer math can be more than 3x faster than Hitech's (see the [pi benchmark](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/pi)) if you're willing to pay the increased code size.  I think making use of this integer math could lead to a very fast float package.

Everyone wants IEEE so doing that at the same time may be the way to go.

Anyway this is a reasonably big project so it will likely fester here for a while.

A

feilipu, 2017-03-20:
It is certainly an interesting and worthwhile challenge.

I'm quite interested to help with this, but not sure that I have enough experience to be able to lead it.

Simply researching the different 32bit formats, and how to convert between them effectively is sufficient for me to chew currently.

suborb, 2017-04-01:
Here's the Rabbit implementation (mostly IEEE compliant):

https://github.com/digidotcom/DCRabbit_9/blob/8bb0fa6037a806acec020ac0014bc059cbf69491/Lib/BIOSLIB/MUTILFP.LIB

https://github.com/digidotcom/DCRabbit_9/blob/8bb0fa6037a806acec020ac0014bc059cbf69491/Lib/MATH.LIB

feilipu, 2017-04-01:
I haven't done a diff on the files (which I will do shortly), 
but is there any other reason to reference Release 9 and not Release 10?

https://github.com/digidotcom/DCRabbit_10/blob/master/Lib/Rabbit4000/BIOSLIB/MUTILFP.LIB
https://github.com/digidotcom/DCRabbit_10/blob/master/Lib/Rabbit4000/Math/MATH.LIB

suborb, 2017-04-01:
No reason. I wasn't even looking for fp routines I just ran across them and didn't check to see if there were later versions. 

I guess later versions may have more of the rabbit specific instructions in them.

aralbrec, 2017-04-02:
Release 10 is for the Rabbit 4000,5000,6000
Release 9 is for the Rabbit 2000,3000

The latter will be closer to z180 but it doesn't hurt to look at both.

feilipu, 2017-04-02:
The MUTILFP differences are small. Just leading underscores, and a few JP for JR exchanges.
There's a few Rabbit specific OpCodes. `EX DE, HL'` instructions scattered through them.

zx70, 2017-11-29:
I'm not sure about it,  so I'm asking here: should the old rom based fp functions be fixed in some way?
They all seem broken now, eg the zx ones crash with an 'invalid argument' basic error.
I'd like to fix all of them but I'm not understanding what exactly changed.

aralbrec, 2017-12-04:
They should still work as-is.

PaintedBlck, 2018-06-20:
Hi,

I’m willing to invest time in this issue. Measuring compiler efficiency has always been a favourite pass-time. As has been optimizing code and pushing performance to the edge.

In order to be able to focus on the issue and not on the entire dev kit, I think it’s better to isolate the actual fp routines (and the ones I develop) from the dk. Not in the least because this will permit to profile, time and compare timings of the real calculations (actual and new), and exclude side-effects.

In order not to loose time with searching, can someone point me out where the current fp routines are? Is there already ieee 32-bits code, or are the 48 bit fp operations all that exist for the moment.

Do I understand well that the goal is to be as fast as Hitech C v309, but also to have solutions for exp, sin, cos, ... ?

Kind regards.

aralbrec, 2018-06-21:
There are two floating point implementations, both 48-bit with 40-bit mantissa.  One is called math48 and the other genmath.  Math48 is faster but genmath can be used with targets that have restrictions on what registers can be used.  sccz80 in the classic library can also interface with a couple of native float implementation on some targets, for example the zx spectrum float implementation in rom and the zx81 float implementation in rom.  But the compiler's float packages are math48 and genmath.

The math48 implementation can be found here:
https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/math/float/math48

aralbrec, 2018-06-21:
Facebook has me pressing ctrl+enter constantly :package: 

Anyway continuing:

The MATH48_MANUAL.txt file is the original documentation and source.  MATH48_README.txt lists some of the changes we made.  In particular, math48 keeps the 48-bit float in registers BCDEHL.  We changed the primary float accumulator to be in the exx set BCDEHL' and the secondary one in the main set BCDEHL.  This way return values are in the exx set which keeps it away from the compilers' own code generation.

When writing asm code for the compilers we try to keep to one index register ix and if you can stay away from af' (and I think hl') you will be able to reach more targets as some have restrictions on register use for hardware reasons.  But if that's too much of a burden (and it may be) I'd leave those targets to genmath.  Sticking to just ix is helpful for zsdcc though.

In the link above, the math48 original (and modified) code is in the z80/core subdirectory.  Functions are prefaced with "mm48_*".  The z80 subdirectory contains the asm interface to the library and is defined more in terms of what math.h is expecting.  Functions here are prefixed with "am48_*" and these are the function asm programmers would use and are the functions the c interface jumps into.  The c interface is in the c subdirectory which contains c/sccz80 for the sccz80 interface and c/sdcc_iy for the primary zsdcc interface.  These functions have prefix "cm48_*".

The c interface is specific to the compiler.   zsdcc only supports a 32-bit nearly ieee format float type.  So since the library is 48-bit, each call into the c interface has to convert the 32-bit zsdcc type to math48 form and each exit from the c interface has to do the reverse.  Anything stored in memory is kept in compiler form.  For zsdcc the 32-bit float is passed around in dehl like 32-bit integers are.  sccz80 supports a 48-bit float natively that is slightly different from math48's so there is a small amount of conversion going on.  And then both these compilers have a set of primitives (function names usually contain a trailing p in the prefix) that are called by the compiler to carry out things like add, subtract, multiply, divide, etc.  Anyway I would ignore all this and concentrate on doing a good asm interface; the c interface can follow.  The support of different float types in the compilers themselves has been mused about but you never know when or if it comes.

The newlib is written to support different floating point formats.  There are some floating point related functions in stdlib:

https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/stdlib/z80

Things like strtod, dtoa, etc.  These functions are also used by printf to print floats and eventually scanf will use them to scan floats (right now scanf in the newlib does not do floats and it's recommended to scan a string instead and atof that).  They expect certain named functions to exist in the math library to help implement their functionality.  I can't list them off the top of my head but this can also be added at the end.

The selection of which math library to use in the new lib is made on the compile line.  "-lm" is currently connected to math48.  The "lm" directory contains the connections to the library and programmer:

https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/math/float/math48/lm

All that is here is the creation of aliases for the math48 functions.  Standard function names are used and pointed at the math48 namespaced ones.  When the user adds "-lm" to the compile line, the compiler and programmer's calls made to the standard names are redirected to math48's implementation.

aralbrec, 2018-06-21:
So primarily I would concentrate on doing a good asm interface and the rest will fall out of that.  Keep in mind trying to stick to only one index register and how values are passed to/from the c compiler / programmer to make the interfacing lightweight.  The choice of keeping a float value in the exx set made by math48 was a good one.  For 32-bit float, using dehl would match how the compilers handle 32-bit ints.  genmath uses some static memory so it is not re-entrant.

The goal would be having an ieee-compliant type but also we'd like to improve performance to try to match hitech c.  Hitech c has a deficient float type so having a working float package would be preferable to matching its speed :)  z88dk also contains some fast integer multiplies that do loop unrolling and leading 0 bit detection that can be selected by the user so if those are useful you could allow the user to determine if he wants to trade some space for speed.

PaintedBlck, 2018-06-21:
Thank you for elaborating. I accept the challenge ;-)

I'll start studying in detail what you have explained during the weekend, and hope to start setting up a test environment next week, so to test the actual code bases you've mentioned.

I recall having used and looked into 8-bit architecture and 16-bit architecture (8086) Fortran 77 IEEE 32-bits floating point libraries that were lightening fast. I'll have a look at those to, to get some extra inspiration.

In short (for my good understanding, please correct me if I'm wrong):
* first priority is getting it to function, second is to speed it up
* stick to one ix and if possible stay away from af' and hl'
* stick to exx when possible (bcdehl')
* first the asm interface, afterward the c interface
* make the lib reentrant
* zsdcc passes around float32 as int32

PaintedBlck, 2018-06-24:
I'm runing in a number of problems when setting up an environment to test, debug and profile my float32 library entirely within the realm of the available z88dk tools.

Where can I describe these problems to get a reply in order to be able to start my real development.

For the moment, I've placed my routines and problem descriptions in a temporary GitHub repository: [https://github.com/PaintedBlck/float32](https://github.com/PaintedBlck/float32)

suborb, 2018-06-25:
Problem 1:

Ticks only runs with the classic library at the moment, so you'd need to compile with +test and then it becomes runnable.

For implementing a maths library I don't think newlib vs classic is massively different if you copy the math48 model.

2 and 3 are possibly related to the classic/newlib problem as well.

Problem 4 is just a pain, if you use sccz80 then the -gcline will help

aralbrec, 2018-06-26:
ticks should work fine as it shouldn't have a problem with where the raw binary comes from.  But you do have to note that ticks expects (or expected at that time - ticks has changed since then!) a memory image starting at address 0.  You'll note in the page you linked to that it mentions the program org was moved to 0 to accommodate that.

aralbrec, 2018-06-26:
I can have a another look tomorrow but for now I'll note that if the c errors happen at link time (eg label not found) then the linker reports the line number in the asm file translation from c.  It doesn't have the c file at this point, only the translated asm.  sccz80 can insert line number directives in its generated asm to fix this but zsdcc cannot.  You can see the asm file translation by adding --list to the zcc line and you can use this to look up the line number for linking errors.

feilipu, 2019-05-03:
> https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/math/float/math48/lm
> All that is here is the creation of aliases for the math48 functions. Standard function names are used and pointed [...]

@aralbrec where is the decision tree to redirect for `-lm` ? Or more specifically, for #1121 I'd like to create another linkage option for (say) `-lm32`. Would it be necessary to patch both compilers to recognise this linkage?

aralbrec, 2019-05-05:
It is fairly straightforward (aside from the volume of work).

Using math48 as an example:

feilipu, 2019-05-21:
Now in #1155 now that math32 has been merged.

closed 2019-08-27 by feilipu