z88dk/z88dk #124: genmath: outlier poor result for spectral-norm benchmark

aralbrec, 2017-03-22:
The [spectral-norm](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/spectral-norm) benchmark is showing an abnormally slow time for sccz80/genmath.

```
5.
Z88DK March 18, 2017
sccz80 / new c library
4026 bytes less page zero

error: 2 * 10^(-9)

cycle count  = 9271618782
time @ 4MHz  = 9271618782 / 4*10^6 = 38 min 38 sec

48-bit float implementation causes relative slowdown.

6.
Z88DK March 18, 2017
sccz80 / classic c library
4106 bytes less page zero

error: 2 * 10^(-9)

cycle count  = 15385384430
time @ 4MHz  = 15385384430 / 4*10^6 = 64 min 06 sec

48-bit float implementation causes relative slowdown.
```

Number 5 shows results for sccz80/math48 and number 6 shows sccz80/genmath.  For some reason, sccz80/genmath is incredibly slow in comparison.  Normally math48 is only slightly faster than genmath.  I haven't had a chance to see what is going on yet.

suborb, 2017-03-22:
Classic/math48 is down about 908 in my branch with classic/genmath about 1493.

Making eval_A just a return 1.0, then we're at 201 and 239 respectively.

So there's still a significant difference. 

In terms of implementation differences there's the parameter passing to the basic operators that stands out (math48 using exx, genmath using the stack + FA variable), maybe it just comes down to that.

aralbrec, 2017-03-23:
I think the only thing significantly different in that function is how the integer is converted to float.

The [math48 conversion](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math48/z80/am48_double32u.asm) is written to be fast (it tries to reduce the long value to 16-bit if it can and then 8-bit if it can from there followed by a whole byte shift and individual bit shifts in fast registers). 

[Genmath](https://github.com/z88dk/z88dk/blob/master/libsrc/math/genmath/norm.asm) doesn't look terribly slow as it does try to whole byte shift if it can but its shifting is always 32 bit and its one bit shift is comparatively slow.

I tried a representative example and the math48 long to float conversion came out twice as fast as genmath's.  This conversion is in the innermost loop of this benchmark so I think the long to float conversion is probably the problem.

suborb, 2019-05-21:
Running this through hotspots on ticks after doing the optimisation for 16 bit floats, it looks like the issue is actually the division routine.

The following sequence dominates the timings:

```
16799771        184797481               DIV2:
                    push    hl                              ;[0e5f] e5
16799771        251996565                                   push    iy                              ;[0e60] fd e5
16799771        184797481                                   push    bc                              ;[0e62] c5
16799771        184797481                                   push    hl                              ;[0e63] e5
16799771        67199084                                    ld      a,b                             ;[0e64] 78
16799771        67199084                                    exx                                     ;[0e65] d9
16799771        319195649                                   ex      (sp),hl                         ;[0e66] e3
16799771        67199084                                    or      a                               ;[0e67] b7
16799771        251996565                                   sbc     hl,de                           ;[0e68] ed 52
16799771        319195649                                   ex      (sp),hl                         ;[0e6a] e3
16799771        67199084                                    ex      de,hl                           ;[0e6b] eb
16799771        251996565                                   push    iy                              ;[0e6c] fd e5
16799771        319195649                                   ex      (sp),hl                         ;[0e6e] e3
16799771        251996565                                   sbc     hl,de                           ;[0e6f] ed 52
16799771        319195649                                   ex      (sp),hl                         ;[0e71] e3
16799771        235196794                                   pop     iy                              ;[0e72] fd e1
16799771        67199084                                    ex      de,hl                           ;[0e74] eb
16799771        67199084                                    sbc     c                               ;[0e75] 99
16799771        67199084                                    exx                                     ;[0e76] d9
16799771        167997710                                   pop     hl                              ;[0e77] e1
16799771        67199084                                    ld      b,a                             ;[0e78] 47
16799771        218397023                                   ld      a,(extra)                       ;[0e79] 3a a0 13
16799771        117598397                                   sbc     $00                             ;[0e7c] de 00
16799771        67199084                                    ccf                                     ;[0e7e] 3f
16799771        160926727                                   jr      nc,DIV4
```

feilipu, 2026-11-08:
Genmath has been retired from active duty. It is still available for those that want it.

closed 2026-11-08 by feilipu