z88dk/z88dk #125: classic: several issues with zsdcc compiles

aralbrec, 2017-03-22:
The issues I've found can be summarized in one example:  [classic/whetstone.c](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/whetstone/z88dk-classic)

1. The sccz80 compile generates proper printout whereas the zsdcc compile does not implement the % converters so that output is in form "e" etc instead of %e

2. The CLIB_OPT_PRINTF option in general doesn't work for sccz80 and zsdcc.  I've had to remove this option for all classic compiles in the benchmarks so that the library selects a proper converter subset selected by the compiler.  Actually this is probably why (1) is not working for zsdcc.  zsdcc does not look at the converter strings to indicate a printf converter subset.

3. There is a /r/n issue with zsdcc and not sccz80.  sccz80 will properly generate /r/n when /n is received from printf by zsdcc only seems to generate /n and the terminal does not return to column 0.  This is with the zx target.

suborb, 2017-03-22:
I can't reproduce 2 at all with a simple program (modified world.c). It only takes the first pragma definition, so if it's defined multiple times they will be ignored.

I think it might be worthwhile introducing #pragma printf "format string" and #pragma scanf "format string" to make life easier here - the code from sccz80 can be used.

aralbrec, 2017-03-22:
> can't reproduce 2 at all with a simple program (modified world.c). It only takes the first pragma definition, so if it's defined multiple times they will be ignored.

Just add "#pragma output CLIB_OPT_PRINTF = 0x05001001" to [classic/whetstone.c](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/whetstone/z88dk-classic) .  It doesn't matter if you compile with sccz80 or zsdcc (per instructions in the readme) - both will print "ld" and "e" instead of numbers.

> I've added #pragma printf "format string" and #pragma scanf "format string" to zpragma to make life that little bit easier.

It's a good idea.

suborb, 2017-03-22:
> Just add "#pragma output CLIB_OPT_PRINTF = 0x05001001" to classic/whetstone.c . It doesn't matter if you compile with sccz80 or zsdcc (per instructions in the readme) - both will print "ld" and "e" instead of numbers.

Yeah that makes sense, flags are 0x40000000, so the option needs to be  0x45001001

The core of the classic printf is this:

    handle_percent:
          ld      a,(hl)
          inc     hl
          cp      '%'
          jr      z,print_format_character
          call    __printf_get_flags              ;level2
          res     6,(ix-4)
          cp      'l'
          jr      nz,no_long_qualifier
          set     6,(ix-4)
          ld      a,(hl)
          inc     hl
    no_long_qualifier:
          push    hl      ;save fmt
     ; Loop the loop
          ld      hl,__printf_format_table
          ld      c,a
    formatloop:
          ld      a,(hl)
          and     a
          jr      z,no_format_found

    ....

    no_format_found:
          ; No matching format character, just print it out
          pop     hl
          ld      a,c
          jr      print_format_character

With out 0x40000000, __printf_get_flags is basically a ret which doesn't swallow the flags in the format, and hence you end up falling through to just printing the format string (which is what you are seeing).

suborb, 2017-03-24:
1. is covered with 2.

I think 3. might be because +zx hasn't switched over to -standard-escape-chars

aralbrec, 2017-03-26:
> Without 0x40000000, __printf_get_flags is basically a ret which doesn't swallow the flags in the format, and hence you end up falling through to just printing the format string (which is what you are seeing).

Ok I see this:

```
IF CLIB_OPT_PRINTF & $40000000
	EXTERN	__printf_get_flags_impl
	PUBLIC	__printf_get_flags
	defc	__printf_get_flags = __printf_get_flags_impl
ELSE
	EXTERN	__printf_get_flags_noop
	PUBLIC	__printf_get_flags
	defc	__printf_get_flags = __printf_get_flags_noop
ENDIF
```

So you're saving some space for selection of simple stdio?

I'm wondering if we can reverse the sense of that $40000000 bit so that is set when simple stdio is selected.  This way there would be no incompatibility with the new c lib.

aralbrec, 2017-03-26:
> I think 3. might be because +zx hasn't switched over to -standard-escape-chars

Yes if you do the sccz80 compile with "-Cc-standard-escape-chars"  you get the same output.

aralbrec, 2017-03-30:
I've updated scanner.c to make the printf and scanf format specifiers more complete.

The scanning in parse_format_string() for classic doesn't look right to me.  It's only setting that 0x40000000 bit if there is a flag or digit in the format specifier.  I would think that that bit has to be set all the time or else the converters specified will be ignored in classic.

suborb, 2017-03-30:
Thanks for finishing that off, that dropped off my radar.

I'm not quite sure I understand your second point though. In classic (for example) a plain %d doesn't need 0x40000000, but %2d does need it. There's some zcc_opt.def magic to merge all the values together to get the complete bitset.

aralbrec, 2017-03-30:
My bad, it's working now.  I think I had some in-between binaries when I was trying this out a few days ago.

closed 2017-03-30 by aralbrec