z88dk/z88dk #129: sccz80: incorrect shift amount in 32-bit unsigned shift left

aralbrec, 2017-03-25:
Test program:

```
int g_rand(void)
{
   static unsigned long s = 0x80000001;
   unsigned long        t = s;
   
   t ^= t >> 10;
   t ^= t << 9;
   t ^= t >> 25;

   s = t;

   return(t & 0x7fff);
}
```

zcc +zx -vn -a test.c --c-code-in-asm
reveals this snippet:

```
;t ^= t << 9;
	LINE	8
	ld	hl,0	;const
	add	hl,sp
	push	hl
	call	l_glong
	push	de
	push	hl
	ld	hl,6	;const
	add	hl,sp
	call	l_glong
	ld	a,4
	call	l_long_aslo
	call	l_long_xor
	pop	bc
	call	l_plong
```

Notice the shift amount loaded into A is 4 instead of 9.

suborb, 2017-03-25:
Introduced by PR #126.

suborb, 2017-03-25:
There's some interesting divergences in undefined behaviour....

I've made sccz80 and libraries match the behaviour of (my) host machine, i.e. effectively do an %32 on the shift count. This means that the libraries match the behaviour of the internal constant folding within sccz80.

sdcc on the otherhand does what I did initially, if the shift > 31 then assign to zero, so doesn't match the constant folding. test_lshift.bin fails on this line:

     long val  = 1;
     Assert( val << 32 == 1L << 32, "<<32"); // Undefined behaviour

but passes successfully on this:

     long val = 1;
     int v = 32;
     Assert( val << v == 1L << 32, "<<32"); // Undefined behaviour

Which is unexpected (despite paddling in the great undefined behaviour ocean)

suborb, 2017-03-31:
Closing this one off, it's undefined behaviour so has to be livable with.

commit e65c0e70 (2017-03-25): Fix #129: loadargc divides by two so of course that went wrong!
commit d83279d7 (2017-03-25): Fix #129: Match undefined behaviour for other platforms
commit 47226453 (2017-03-25): Fix #129: Some tests for behaviour (including undefined)
commit a504a04c (2017-03-25): Fix #129: Missing break in the 16 bit case, and actually make the tests
test that.

Add in a warning (it's duplicate printed, but...)
closed 2017-03-31 by suborb