z88dk/z88dk #130: zsdcc: Optimisation error removing valid code?

suborb, 2017-03-25:
Ironically this is in the unit test framework.

Tests generally fail reporting a problem in the setup function, even when there's no setup defined.

Go to test/suite/sccz80 and run 

    zcc +test -vn -I../../framework -Wn9 -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c lshift.c -o test_lshift.bin -lndos -compiler=sdcc
    ticks test_lshift.bin

This will fail due to the comment on #129, however it will report this:

    Starting suite Left shift Tests (4 tests)
    Running test test_lshift32_const.....failed (in setup)lshift.c:185 (<<32)
    4 run, 3 passed, 1 failed

Now, look at the assembly and --c-code-in-asm, lines 61 and 64 (the stage increment) have vanished.

We're also getting this warning:

    ../../framework/test.c:77: warning 110: conditional flow changed by optimizer: so said EVELYN the modified DOG

aralbrec, 2017-03-25:
The EVELYN message comes when code is eliminated, usually from conditional branches that can't occur.

The two stage increments are eliminated but it's right to do that.  The value is never referenced in that branch of the if and at the end of the if, the program loops back where stage is immediately set to zero.  So the stage increments are unnecessary work.

suborb, 2017-03-25:
The problem is that that that branch is within a setjmp(), when/if the longjmp() occurs the stage values are needed.

aralbrec, 2017-03-25:
Yeah I see now -- it can't do that inside a setjmp controlled block.  That would be a bug and probably a difficult one to solve if they haven't thought of that before.

I think there is a way to disable that optimization.  I'll take a look now.  This is something I'd report to sdcc bugs if you want to do that.

aralbrec, 2017-03-25:
I found this:

"All accessible objects have values as of the time longjmp() was called, except that the values of objects of automatic storage duration which are local to the function containing the invocation of the corresponding setjmp() which do not have volatile-qualified type and which are changed between the setjmp() invocation and longjmp() call are indeterminate."

so it's indeterminate behaviour and not a bug.

suborb, 2017-03-25:
Ah, today is my day for finding undefined behaviour!

So making it volatile disables that optimisation, as does making it static.

closed 2017-03-25 by suborb