z88dk/z88dk #133: z88dk: namespacing of i/o ports

aralbrec, 2017-03-26:
In the new c library there are a lot of constants that are defined at library build time that need to be communicated to the user program at compile time.  These are things like i/o port assignments, fixed memory map locations, c library configuration, cpu characteristics, and so on.  These things can change depending on target config.

The method being used to do this has the user modify the configuration in m4 files located in [z88dk/libsrc/_DEVELOPMENT/target/TARGETNAME](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/target).  When the target's library is built these m4 files are processed to generate two monster files that hold all the final defines for the target.  One file holds the definitions for assembly language and the other holds it for C.  The [assembly language defines](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/z180/config_z180_public.inc) are actually exported as part of the target library so that they are accessible in asm via suitable EXTERN declarations.  The [C defines](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/z180/config_z180.h) are held in a header in the target directory.  The example links are for the default build of the z180 target.

For C, this header file is accessible via `#include <arch.h>` which will include it wholesale.  arch.h is a switch on the compiler target that ensures the correct defines are included.

All these defines get poured into the global scope which can conflict with user defined names.  So what still needs to be done is to make sure all these symbols stay within the implementation's private scope, which is a scope defined by the C standard as any name starting with two leading underscores.  So that's part of what will happen next -- all these names will be changed to have two leading underscores.  Then the intention is to expose these names selectively.  So, for example, a "z180.h" header might define all the z180's built in i/o ports.  So z180.h would be primarily concerned with creating new names without the two leading underscores to be equivalent to the private scope names with two leading underscores.  Eg, z180.h would create a new name "IL" equivalent to the already defined "__IL" corresponding to i/o register IL.  It would selectively expose "IL" into the user's namespace.

Other headers might expose other implementation-private config information and the same may be done for assembly language.

That's the long story... the short story is I think i/o ports should be in their own namespace within the user program.  The example I gave for the z180, i/o port "IL", is a very short name that could conflict with user-defined symbols.  Now, on a z180, these short port names are defined by the manufacturer, so maybe the risk of conflict is smaller but there are a lot of targets in z88dk where port names are not well defined.  For example, on the sega master system, there is no agreed upon names for its dozen or so hardware registers.  The same goes for the zx spectrum target, whose problem is compounded by a lot of 3rd party hw whose port names may conflict (what about the FDC register names for floppies on two different devices which use the same FDC controller).

What I'd like to do is preface all port names with "io_" (my favourite prefix at the moment, other ideas being "port_" and "p_").   Where more specific information is recommended, io ports can be further namespaced according to manufacturer (io_SINCLAIR_FDC, "io_OPUS_FDC").   For the z180, I would like to do "io_IL" by default but then a standard header should alias "IL" to "io_IL"; this I think has to happen because Zilog has defined the name "IL".

Anyone have any other preferences or better ideas?

pauloscustodio, 2017-03-26:
I agree.

aralbrec, 2017-03-27:
Ok I've started to adopt this method.

The implementation will define constants for port-mapped and memory-mapped devices with "__IO_" and "__MM_" prefixes.  These names are exposed to assembly language through PUBLIC exports in the library with suitable renaming (usually eliminating the two leading underscores but may also involve using a name defined by the target).  They are exposed to C in the raw through inclusion of "arch.h".  Other C headers will include "arch.h" and selectively place these defines into the user namespace by renaming to eliminate leading underscores or by using a name defined by the target.

closed 2017-03-27 by aralbrec