z88dk/z88dk #134: ticks: need a way to perform simple putchar, getchar and retrieve timer inside simulated programs

aralbrec, 2017-03-27:
Ticks will probably end up being turned into a more comprehensive simulator but in the meantime there is a need to be able to perform character i/o to the terminal and to retrieve the timer value inside the simulated program.

In most of the benchmarks, verification and timing is separated with a verification step using printf run inside a specific target's emulator and then the timing step without printf run using ticks.

With coremark this can't be done.  The rules for coremark require that the source code not be modified (and I'm not sure I want to go down that rabbit hole anyway) and it needs access to both timers and printf.

So let's put it all into ticks.  This will also simplify the evaluation of the other benchmarks, making it all one step.

The idea is to add an option "-iochar=nn" to define an 8-bit i/o port that ticks the program will monitor.  Writes to that port will result in a putchar to the terminal and reads will result in a getchar.

Also add two options "-iotimer=nn" to define an 8-bit i/o port that resets the current count on write and writes the current 8-byte count into memory on read.  The address where this 8-byte count is written is given in option "-mmtimer=addr".

suborb, 2017-03-27:
Ticks can already do console io and I think a second resolution time function. Take a look at the hook*.c files in ticks and stdio/test in the classic library. 

This mechanism might as well be extended for anything more we want.

soster, 2022-08-28:
Yeah the commit is not working. I already found the bug and I am preparing a merge request.

wrong:
`int hook_console_out(int port, int value)
{
    if (ioport !=-1 && (port&0xff) == ioport) {
        fputc(c,stdout);
        return 0;
    }

    return -1;
}`

correct:
`int hook_console_out(int port, int value)
{
    if (ioport !=-1 && (port&0xff) == ioport) {
        fputc(value,stdout);
        return 0;
    }

    return -1;
}`

closed 2026-10-27 by feilipu