z88dk/z88dk #135: Restore Rabbit x000 support

suborb, 2017-03-30:
It ended up being disabled a while back..

Would be good to support it properly, taking advantage of the CPU features. 

* Given the amount of effort that has gone into the peephole rules for sccz80, lets not destroy them by changing the compiler output, use a rule file to do the conversion.
* Adding support to z80asm would be useful but not essential.
* Updating ticks to support the rabbit instruction set would be really useful as well.
* Re-enable the library building

suborb, 2017-04-08:
Sdcc can now be used to compile rabbit programs. It uses the -mr2k flag to set the cpu type. 

That flag is also supported by sccz80 which uses it for stack cleanup. Further rabbit optimisations are performed by a cpu specific rules file. 

The rabbit instructions are then translated into a defb sequence before assembly.

suborb, 2017-04-10:
@pauloscustodio what are your thoughts on adding the R2K instructions to z80asm?

pauloscustodio, 2017-04-10:
I agree to put it in the to-do list.

I need to have a look at the instruction set to check if it is still
feasible with the current z80asm.

suborb, 2017-04-11:
Thanks Paulo. I've made a start at it, and it looks feasible so far (subject to the caveat that I've not  done any ld instructions yet).

The data sheet I'm using as reference is this one: http://datasheets.chipdb.org/Rabbit/RabbitInstructions.pdf

pauloscustodio, 2017-04-12:
It is feasible in the current z80asm. I will add a new option --cpu=r2k (and remove the old --RCMX000) and implement the new opcodes.

pauloscustodio, 2017-04-12:
I've seen the work @suborb started on feature/z80asm_rabbit. Looks good, thanks.
Shall I take over?

pauloscustodio, 2017-04-12:
I have added the Rabbit documents to https://github.com/z88dk/techdocs and also updated z80_cpu_um80.pdf to the 2016 version.

suborb, 2017-04-12:
Thanks again Paulo, once I figured out how to regenerate the parser it was pretty easy to extend.

 I think we probably need to add 3 command line options:

--cpu=r2k
--cpu=r3k  (there's a couple of extra instructions)

and another one to turn on the emulation functionality.

Again here I'm not sure about how the testing works.

I thought I'd have the time to finish this one, but Alvin has popped up with a couple of sccz80 issues which I should tackle, so I may not be able to finish this particular one so if you want to take over then feel free.

pauloscustodio, 2017-04-12:
Ok, I will take over.

I will start with adding unit tests and then the code.

What do you mean with "... and another one to turn on the emulation
functionality"?

Thanks.

suborb, 2017-04-12:
Re the emulation. There's a set of instructions: cpir, cpi, rrd etc that we emulate on the Rabbit by calling library routines.

Likewise, we emulate the missing call flags by jumping over them.

pauloscustodio, 2017-04-12:
Got it. IMHO we do not need an option for these - we just accept them and call the emulation routines. In case the user does not link the required library, she will be notified by undefined symbols link error.

suborb, 2017-04-13:
Good point, if they're actually writing true Rabbit code they'd not use those opcodes, as long as it fails that's what's important.

pauloscustodio, 2017-04-17:
With the commit e29bee72864b1fed6cb79a186567205998776ec5 in the feature/z80asm_rabbit branch I have removed the --RCMX000 option to z80asm and added instead the --cpu=r2k and --cpu=r3k options.
I have also added a --no-emul option to issue an illegal identifier error message instead of calling library routines for opcodes not existing in the Rabbit.

The main build is now failing at 
```
zcc +rcmx000 -O3 -vn -Wn43  -c -o obj/r2k/regexp.o -c regexp.c
Error: illegal option '--RCMX000'
1 errors occurred during assembly
```

Can I change zcc to use +r2k instead of +rcmx000, and do the same in the call to z80asm?

suborb, 2017-04-17:
Yup, change lib/config/rcmx000.cfg to refer to the new option.

suborb, 2017-09-03:
The +test target now has a -clib=rabbit target to use Rabbit libraries.

ticks now emulates the instructions that we use, z80asm fully supports the Rabbit CPU. 

Optimisation rules have been added to use the Rabbit rules, some of the libraries have been optimised.

closed 2017-09-03 by suborb