z88dk/z88dk #136: classic: text terminal discrepancy in sdcc versus sccz80

aralbrec, 2017-03-30:
As noted elsewhere there is a /r/n incompatibility issue in the libraries with zsdcc versus sccz80.  The particular test case is the zx target.

If you compile using zsdcc for the zx target /n causes the text cursor to move downward but not to column 0.  However compilation with sccz80 will do both things.  This is because zsdcc uses correct ascii encoding for /r/n whereas sccz80 reverses that.  This decision was made long ago to simplify terminal drivers when we thought most targets used code 13 to to /r/n.

You can get the same behaviour from sccz80 if you ask sccz80 to generate standard escape chars -Cc-standard-escape-chars (I think).

The fix is to normalize all the text drivers to use the correct ascii codes for /r/n.

As noted elsewhere there is a /r/n issue with zsdcc and not sccz80.  sccz80 will properly generate /r/n when /n is received from printf by zsdcc only seems to generate /n and the terminal does not return to column 0. This is with the zx target.

suborb, 2017-04-01:
Current status of enabling the \n \r swap round:

* [ ] +abc80 ABC80
* [ ] +abc800 ABC800
* [ ] +ace Jupiter Ace
* [ ] +aquarius Mattel Aquarius
* [ ] +bee
* [ ] +c128 Commodore 128
* [ ] +c7420
* [ ] +cpc Amstrad CPC
* [x] +cpm CP/M
* [ ] +embedded
* [ ] +enterprise Enterprise
* [ ] +gal Galaksija
* [ ] +kc
* [ ] +lambda
* [ ] +lynx
* [ ] +m5 Sort M5
* [ ] +mc1000
* [x] +msx MSX
* [ ] +mtx Memotech
* [ ] +mz Sharp MZ
* [ ] +nascom Nascom 1/2
* [ ] +nc Amstrad NC100
* [ ] +newbrain Grundy NewBrain
* [ ] +osca Old School Computer Architecture
* [ ] +oz Sharp 700
* [ ] +p2000
* [ ] +pacman Pacman Arcade Hardware
* [ ] +pc6001
* [ ] +pp2 Peters Plus Sprinter
* [ ] +rc2014
* [ ] +rcmx000 Rabbit boards
* [ ] +rex Xircom Rex 6000
* [ ] +sam Sam Coupe
* [ ] +sc3000 Sega 3000
* [ ] +sms Sega Master System
* [ ] +sos
* [ ] +srr
* [ ] +svi Spectravideo
* [ ] +test Unit test/ticks target
* [ ] +ti82
* [ ] +ti83
* [ ] +ti85
* [ ] +ti86
* [ ] +ti8x
* [ ] +trs80 Tandy TRS/80
* [ ] +ts2068 Timex 2068
* [ ] +vg5k
* [ ] +vz
* [ ] +x07
* [ ] +x1
* [ ] +z1013
* [ ] +z88 Cambridge z88
* [ ] +z9001
* [ ] +zx Sinclair ZX Spectrum
* [ ] +zx80 Sinclair ZX80
* [ ] +zx81 Sinclair ZX81

suborb, 2017-04-06:
This is slow going, and I don't have emulators for the remaining machines so I'm going to have to stop.

@zx70 have you got working emulators for all the targets? Can I pass this one on to you to finish and verify?

Basic changes per target are:

lib/conf/?.cfg 
- add -Cc-standard-escape-chars

libsrc/Makefile
- add -DSTANDARDESCAPECHARS to build line

libsrc/stdio/?/fputc_cons.asm
- Handle 10 or 13 as the linefeed depending on STANDARDESCAPECHARS (I think a lot of them are actually done already)

Verify with a console app.

zx70, 2017-04-07:
I think I can do it,  I try and let you know.

zx70, 2017-04-08:
Iid propose to rename clib.lst  to *_legacy.lst  and replace it with the new lst  version.   Opinions!?

suborb, 2017-04-08:
Naming them _classic will help with the library reorganisation. 

I'm thinking of doing a bit of zcc work to make new lib a first class citizen in the config files. The subtype lines work but are a bit hairy.

zx70, 2017-04-09:
Good. While I'm on the cfg files, is it already feasible to add an option for sdcc and classic lib?

suborb, 2017-04-09:
I'm not sure it's worth it for classic. To switch to sdcc it's just a case of adding the -compiler=sdcc option which is probably as much typing effort as adding a subtype= option!

zx70, 2017-04-09:
wonderful!

zx70, 2017-04-10:
I moved "stdio_stdesc.lst" to "stdio.lst" and made it become it to default.
The old .lst file is now renamed to "stdio_classic.lst" ..perhaps the Rabbit 🐰 target should now to use it ?

"stdio_stdesc.lst" does no longer exist, hope it won't be a pain

suborb, 2017-04-10:
I think I need to do a dedicated build of everything for Rabbit. Though I'm running a bit blind since I've not got a board and I guess mimarob has moved on.

zx70, 2017-04-11:
Trying to balance our energies (to keep this activity in a reasonable amount of time) I will try to do a quick pass of all the target with classic z88dk library support and open specific issue tickets for the single problems which can be fixed on the fly.    This approach should permit a new release to happen soon and all those targets easily fixable to be ready first.

zx70, 2017-04-12:
I could reproduce a problem only today:  some of the converted targets can't run Othello properly:
   while (c != '\n' && c != 4) c= getchar();
..where getchar() is probably an alias for getk().   Need to double-check it, the bug makes the text input never terminating.   Spectrum works well.

zx70, 2017-04-13:
I'm stuck with on the input part.
A minimalistic fix like the following one is not enough:

> IF STANDARDESCAPECHARS
> 	cp	10
> 	jr  nz,esc_switched
> 	cp	13
> 	jr  nz,esc_switched
> 	xor 7
> .esc_switched
> ENDIF

Completing the LF/CR  sequence by buffering the CR character for the next read doesn't seem to be enough.
Ideas ?

zx70, 2017-04-21:
I'm stopping here.   Can you close the few missing ones ?

suborb, 2017-04-21:
Wow, thank you. Lets try not to have one of those big behavioural changes again.

Yup, I'll clear up the dregs.

zx70, 2017-04-22:
It has been a good opportunity to revise the ports.   A couple of them were not functioning anymore, eg those not bound to appmake (Amstrad NC.. I had to port the emulator to Windows, first!)  missed the data preload.

suborb, 2017-04-22:
Did you manage to sort out the Othello problem?

zx70, 2017-04-22:
No, sadly the input part is still broken on all the targets.
I can probably adapt Othello but it'd be cheating.

suborb, 2017-04-22:
Okay, I'll take a look and see how we can fix it.

suborb, 2017-04-23:
I tried the mapping on the ZX fgetc_cons() and it works as expected so Othello is repaired for that port so I think we need to go through all the ports again :(

aralbrec, 2017-04-23:
I think before anything is done test on a target that sends \r\n on the input stream (cpm?)

```
+IF STANDARDESCAPECHARS
+	cp	13
+	jr	nz,not_return
+	ld	a,10
+.not_return
+ENDIF
```

For a \r\n input stream, that's going to generate two carriage returns.
In the new c lib I am swallowing all \r so that on the c lib side there is only \n to deal with.

zx70, 2017-04-24:
>For a \r\n input stream, that's going to generate two carriage returns.
>In the new c lib I am swallowing all \r so that on the c lib side there is only \n to deal with.

Right.  Anyway this can be done locally on those few targets sending multiple control codes.
I can do it.

zx70, 2017-04-24:
I reset the list above, and start the second run  🏃

zx70, 2017-04-25:
Sorry for asking again, but I noticed the Spectrum port was fixed only on fgetc_cons().
I thought also getk() had to be lined-up to avoid confusion with the control codes, but honestly I'm not 100% sure.   Demos like "wintest.c" can be quickly fixed, what's the best option in your opinion ?

suborb, 2017-04-25:
I wasn't sure originally, I fixed up getk but didn't commit it. 

I was musing just leaving it as is, but that reduces portability. However that reduces portability so I think it needs to be changed as well.

zx70, 2017-04-25:
Good.   No worries, most of the input routines are so close relatives that most of them barely needs to be tested.

zx70, 2017-04-25:
Put a flag also on X07, it is far too experimental to pass an input test.

zx70, 2017-04-26:
The problem of extra characters does not seem to be serious.   Few console drivers were already generating extra characters (on OSCA I did it by purpose) and all the programs I could test happily leaved with it.
We now have a minor conflict for the cursor key codes: I used the ZX Spectrum encoding here and there and CHR$(10) is used also for "cursor down" (e.g. on the TI calculators), now ENTER and cursor down are now producing the same key code.   It could be a problem only in those cases where "cursor down and ENTER" are both required.

zx70, 2017-04-26:
Again, I'm stopping here, please have a look at the remaining ones  :)

aralbrec, 2017-04-26:
There are no issues in the new c lib so I've checked off the rc2014.

suborb, 2017-04-30:
Rex doesn't have any stdio, embedded, rabbit just use a serial port.

suborb, 2017-09-16:
Just chasing old issues, +cpc was updated.

closed 2017-09-16 by suborb