z88dk/z88dk #138: sccz80: unable to initialize auto variables

aralbrec, 2017-03-31:
sccz80 almost compiles a large sample program without intervention.  It's hanging up on an initialization of an auto variable:

```
void main(void)
{
	const char string[] = "Hello";
}
```

It could be implemented for arbitrary structures by reserving space on the stack and then doing a memcpy from the initialization data stored in the compiler's rodata section.

sdcc emits some terrible code for this:

```
	ld	l,(ix-2)
	ld	h,(ix-1)
	ld	de,0x0004
	add	hl,de
	ld	(hl),0x53
	ld	l,(ix-2)
	ld	h,(ix-1)
	ld	de,0x0005
	add	hl,de
	ld	(hl),0x20
	ld	l,(ix-2)
	ld	h,(ix-1)
```

That's two characters written to that stack variable.  It's a regular pattern though so maybe something can be done in the peephole optimizer.

suborb, 2017-03-31:
Is that sdcc initialisation even right? That snippet looks like the initialisation of a char *string[] with the picking up of a pointer from (ix-2). examples/z88/define.c has a few examples of how these are handled in sccz80.

With sections available it should be a bit easier to handle this now so I'll take a look at it.

aralbrec, 2017-03-31:
That was actually from another compile.  sdcc created a temp variable at (ix-1)/(ix-2) to hold the address of the beginning of the string.  And you can see for every char, it retrieves the start addresses, adds an offset and then writes one byte.

In this specific example sdcc is doing this:

```
	ld	hl,-6
	add	hl,sp
	ld	sp,hl
	ld	hl,0x0000
	add	hl,sp
	ld	(hl),0x48
	ld	c,l
	ld	b,h
	inc	hl
	ld	(hl),0x65
	ld	l, c
	ld	h, b
	inc	hl
	inc	hl
	ld	(hl),0x6c
	ld	l, c
	ld	h, b
	inc	hl
	inc	hl
	inc	hl
	ld	(hl),0x6c
	ld	hl,0x0004
	add	hl,bc
	ld	(hl),0x6f
	ld	hl,0x0005
	add	hl,bc
	ld	(hl),0x00
```

It creates space for the string and again pokes one byte at a time.  This time it holds the start address of the string in BC instead of in a temp on the stack.

It's quite poor and it seems to be something that should be easy to fix inside sdcc.  Maybe I'll see if Philip has time to look.

suborb, 2017-04-01:
Merged in pr #142

closed 2017-04-01 by suborb