z88dk/z88dk #145: new c lib: polish sms target

aralbrec, 2017-04-05:
There are a few loose ends associated with the new c library's sms target.

* [ ] add optional printf driver.  What tile definitions should be used??
* [ ] verify aplib decompression library works properly.
* [ ] verify zx7 decompression into vram works properly.
* [ ] ensure the memory mapped registers MM_FFFF, MM_FFFE, MM_FFFD, MM_FFFC are available through sms.h
* [ ] ensure the vram <-> memory transfer functions are available from sms.h and working
* [ ] verify z88dk/devkitSMS operates properly #57

aralbrec, 2017-04-07:
The devkitSMS integration is complete.

There are several examples committed into the `newclib/IncorporateDevkitSMS` branch that have not been merged into `master` because I haven't received permission to host here.  I've made them available for testing with sample compile lines listed in "readme.txt" files.

The candy kid demo does not behave like the .sms file supplied in the source archive.  It looks like what is compiled is a debug mode compile so I will have to look into that further to make sure something else isn't going on.

The space hawks game cannot be compiled at SO3 level which means a bug has been discovered in the aggressive peephole set.  Fun times ahead locating that one... The game will compile properly at SO2 level.

sccz80 can't be used to compile most of these programs because of lack of support for multi-dimensional arrays and it's missing a feature to change output section names, eg for changing which banks code and data are assigned to.  Something we have on our to-do list is to properly implement a bank assignment solution, of which sdcc only currently solves a small part.

suborb, 2017-04-09:
> sccz80 can't be used to compile most of these programs because of lack of support for multi-dimensional arrays and it's missing a feature to change output section names, eg for changing which banks code and data are assigned to. Something we have on our to-do list is to properly implement a bank assignment solution, of which sdcc only currently solves a part.

Multi-dimensional arrays feels like it's next on my list to do then....

When you say changing section names, is that on the command line? If so then that should be a fairly easy win to add that functionality.

aralbrec, 2017-04-09:
> Multi-dimensional arrays feels like it's next on my list to do then....

This is stopping a number of programs from being compilable under sccz80 including in the benchmarks set.  I know this has been regarded as hard to fix in sccz80, maybe more because of rusty familiarity with the source code, but now that you're on a roll, if you can do it, it would be a pretty big step toward C89 compliance.

> When you say changing section names, is that on the command line? If so then that should be a fairly easy win to add that functionality.

Yes.  sdcc has --constseg NAME and --codeseg NAME to change section names from rodata_compiler and code_compiler respectively.

An example for the sms is [moggy master](https://github.com/z88dk/z88dk/blob/newclib/IncorporateDevkitSMS/libsrc/_DEVELOPMENT/EXAMPLES/sms/moggy-master/readme.txt).

The sega master system normally structures its memory map such that the bottom 32k is fixed, the next 16k is bankable and the last 8k is ram (the top 8k is a mirror of this ram section).  Banks 0 and 1 refer to the first two 16k banks in the fixed bottom 32k.  Then they number banks from 2 to 31 that can be mapped into that 0x8000 - 0xbfff range.  The new c library (and almost the classic too since this definition is carried into appmake) defines BANK_02, BANK_03, ..., BANK_1F sections to hold code or data that is bankable in that address range.  If user code or data is assigned to those banks, appmake will properly form a .sms file holding that code or data properly.

The compile sequence for moggy-master for z88dk and zsdcc is:

zcc +sms -vn -c -clib=sdcc_iy --constsegBANK_02 bank2.c
zcc +sms -vn -c -clib=sdcc_iy --constsegBANK_03 bank3.c
zcc +sms -vn -startup=8 -clib=sdcc_iy -SO3 --max-allocs-per-node200000 wtf.c bank2.o bank3.o -o wtf -create-app -pragma-include:zpragma.inc

It's defining const data for banks 2 and 3 and the "--constseg" switches get zsdcc to change the name of the rodata section from "rodata_compiler" to "BANK_02".   Then this data will be properly assigned to section BANK_02 which is defined by the [sms crt](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/sms/crt_memory_map.inc).

sdcc is limited in what it can do.  Although you can also change the code section to assign code to another bank, you can't do the data section or the bss section.  It's also at a file granularity so you can't selectively assign stuff inside a c file.

In z88dk you have exact control from assembly language by assigning things to specific SECTIONs in asm files but all the tools people write for sdcc target c only because sdcc has weak support for supporting mixed asm and c.

So yes we can do a quick win for sccz80 to be compatible with zsdcc by adding the constseg and codeseg compile line switches.

A deeper solution will allow pragmas to be embedded in c source to switch code/rodata/data/bss sections inside c source files and allow default code/rodata/data/bss sections to be defined on the compile line (which sdcc does for rodata/code).

aralbrec, 2017-04-21:
The remaining problems with sccz80 compiles in some of the devkitSMS examples are due to bugs reported elsewhere or features missing (multi-dim arrays, typedefed function pointers, declared variable in for-loop initializer).

The Astro Force game as example in [libsrc/_DEVELOPMENT/EXAMPLES/sms/astro-force](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/sms/astro-force) is an especially  worthwhile game to check out.

closed 2017-04-21 by aralbrec

zx70, 2017-04-21:
Impressive work,  well done guys !