z88dk/z88dk #149: classic: v6z80p (OSCA) argv wildcards etc..

zx70, 2017-04-09:
After the major change on the startup code, the ddir example and the similar programs don't behave correctly in browsing the directory etc.. I fixed the cp/m equivalent (command line is reversed and only few cp/m  implementations support this option), but I couldn't find the problem on OSCA yet.

aralbrec, 2017-04-16:
Does this change affect the wildcard parsing:
- [all] #105: main() always takes its parameters in r->l format regardless of compiler

zx70, 2017-04-19:
Just checked DDIR, the wildcards work correctly but the file redirection fails, probably this is still connected to parameter order.

suborb, 2017-11-27:
It looks like I didn't update crt0_command_line.asm when I added another member to the FILE structure.

As a result, handling stdout indirection was broken.

zx70, 2018-04-15:
I tested it, but something does not seem to work.  The output file is opened, then the os reboots.
I'll check the osca specific fcntl functions to understand if they were compromised by the lib evolution.

suborb, 2018-04-16:
There's a problem within fputc - I found it whilst working on another issue.

suborb, 2018-04-17:
The callee stack equalisation code was being added at the end of  fputc_callee() - the function wasn't marked as __naked

feilipu, 2026-11-08:
Fixed in 11cc7fec89 (`Issue #149: fputc_callee was C coded compiled as callee so it cleared up its stack automatically.`).

That commit removed the extra callee stack cleanup that rebooted OSCA after `freopen`. `74ac15be73` only fixed the FILE offset, and the failure still reproduced after it. The wrappers are now `__naked`, and stdout is redirected at `__sgoioblk+10`.

closed 2026-11-08 by feilipu