z88dk/z88dk #153: Should we be doing another release soon?

aralbrec, 2017-04-09:
The list of changes is getting very lengthy.  A lot has happened in sccz80 especially.  I think we should consider doing a 1.99C release to keep the list manageable even though we are far from December 31.

If everyone agrees I think before that happens some things should be completed (feel free to add to it):

* [ ] new c lib sms target completed (printf and a few minor items remain)
* [ ] sccz80 SFRs completed
* [ ] sccz80 --constseg and --codeseg compatibility with zsdcc completed
* [ ] sccz80 multi-dimensional arrays (push that for later - this may be longer term?)
* [ ] classic c library transition to correct LF/CR in terminals completed
* [ ] classic RCMXXX support completed
* [ ] new c lib z180 target first pass completed
* [ ] new c lib transition to SFRs for sccz80

Too soon or good idea?

suborb, 2017-04-10:
Yes, this moving to GitHub thing really did give us a bit of impetus! 

Probably best to drop multi-dim arrays out of the candidate features, I've not had a chance to look at how to do it yet, but my gut says that I'm not going to enjoy it that much!

I'm all for a new release, I've been roughly keeping changelog.txt up-to-date and it's already quite long with just the classic and sccz80 changes.

pauloscustodio, 2017-04-10:
I agree with a release.

The new z80asm is still a long way to complete.

zx70, 2017-04-10:
I understand it will take a bit more but I'd like to have the classic lib adjusted as for #136 . 
It will permit sdcc to be used with most of the targets:  adjusting the code is not a long work but testing/revisioning all the targets takes more than expected.

suborb, 2017-10-29:
@aralbrec given the time of year, I guess it's almost time for another christmas present?

aralbrec, 2017-10-29:
Yeah it looks like we missed the mid-year deadline.  I think whatever we have for December will be it.

feilipu, 2019-02-11:
1.99c ?

closed 2019-02-11 by suborb